ARISE-Initiative/robosuite #6: Velocity limits for Sawyer model

tkelestemur, 2018-11-14:
I was wondering how you came up with the velocity limits and `kv` value for the velocity actuators that is defined in Sawyer [model](https://github.com/StanfordVL/robosuite/blob/master/robosuite/models/assets/robots/sawyer/robot.xml). I've checked the real robot velocity [limits](https://github.com/RethinkRobotics/sawyer_moveit/blob/master/sawyer_moveit_config/config/joint_limits.yaml) but they don't match with your values. Also, how did you set `kv` gains for the mujoco ?

amandlek, 2018-11-15:
We basically did some testing for what values worked well with MuJoCo. We don't have guarantees on the simulation parameters used in our repository matching those in the real world. Doing this sort of system identification for simulation parameters that work well to match with the real robot can be an involved process, and we have not done this yet.

closed 2018-11-15 by amandlek

tkelestemur, 2018-11-15:
I see, thanks! I was trying to tune these values as well and I can confirm these worked best.