ARISE-Initiative/robosuite #14: A Question about Human demonstraions

isk03276, 2019-01-06:
Hi. I would like to ask about the demo data provided.

I wonder what elements are contained in "states (dataset) - flattened mujoco states, ordered by time".

amandlek, 2019-01-11:
What exactly do you mean by the rendering screen being turned back on when resetting?

I've seen the GLFW missing error when trying to use on-screen rendering on its own (the fix is to run `export LD_PRELOAD=$LD_PRELOAD:/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/nvidia-410/libGL.so`, assuming you're on Ubuntu with the nvidia-410 driver) but not when using off-screen rendering. Can you post a more detailed example (code and stacktrace)?

Also, we currently don't have support for using both the `has_renderer` option and `has_offscreen_renderer` option, but it should be a pretty easy fix - let us know if you need this functionality and we can address it.

isk03276, 2019-01-14:
Thanks,

"rendering screen being turned back on when resetting" means that the mujoco_py (rendering process) is turned off every episode (whenever env.reset () is executed).

amandlek, 2019-01-14:
Right, this is intentional behavior. We destroy the renderer and create a new one on every environment reset.

closed 2019-01-14 by amandlek

isk03276, 2019-01-14:
If the renderer is turned off and on for every episode, it is difficult to do anything else.
Is there a way not to destroy the renderer?

amandlek, 2019-01-14:
I don't know what you mean by "it is difficult to do anything else" but you can try modifying the behavior [here](https://github.com/StanfordVL/robosuite/blob/master/robosuite/environments/base.py#L146) to avoid destroying the viewer on every reset and see if that is suitable for your use case.