ARISE-Initiative/robosuite #21: Getting the top_site and bottom_site parameters

gsp-27, 2019-02-20:
Hi,

I am trying to load different objects in your amazing environment to train a general purpose grasper. For that I have to provide different xml's for each object. In these xml's there are "site" parameters which specify the topmost and bottommost part of the object, from the center of the object. Also looking at those values they are only computed in z-direction. But when I use the models provided in meshes of this repository and try to compute this top-site and bottom-site, I do not get the same answer as that written in the xml files. So I just find out the mean of the stl file, this is my center and then subtract this center so now everything lies between 0 and 1 and then I just find out the min-max in the z-direction to get the top site and bottom site. 

Maybe I am doing something wrong, any help regarding this topic would be great.

Thanks

jirenz, 2019-02-20:
The 3-D coordinate of the site matters, and the site references body coordinates. Because when we are putting the object **on** the table (say on `table_top = np.array([0.5, 0.5, 0])`, we do `obj.pos = table_top - obj.bottom_offset`. When you have a mesh object, where the bottom should go depends on your mesh. 

@anchit should be able to tell you how the sites for stl files are generated.

gsp-27, 2019-02-20:
Yes. But I am not sure how these values are computed, because when I compute them using the top-most and bottom-most point from the mesh, I do not get the same value as that written in the xml file.

anchit, 2019-02-20:
The top and bottom sites for the mesh objects have not been computed in a principled way. We have just approximately tuned them to what works well. Currently these are only used to initialize at the right place and hence being accurate isn't necessary.

gsp-27, 2019-02-21:
I see thanks for the info.

But is the method that I described above the intended method to compute these parameters, because I am trying to load random shape net objects in the scene and hence it would be beneficial to have these automatically computed.

jirenz, 2019-02-21:
I think it would work. The objects might tilt a bit because the x, y coordinates are not necessarily correct but what you described is what I would do for shapeNET as well.

gsp-27, 2019-02-21:
Okay thanks a lot

closed 2019-02-21 by gsp-27