ARISE-Initiative/robosuite #23: Changing scene/object property on the fly

yuchen93, 2019-03-17:
Hi, I am wondering is there a way to remove/add objects (and also change properties of objects) in a particular environment on the fly?
It seems that this used to be a problem with mujoco-py but they have fixed the read-write issue. How can I do the same in robosuite environments?

jirenz, 2019-03-17:
We do not provide functionalities to remove/add/modify objects on the fly because the xml for the environment is fixed. What you could do is to add all objects, and just "teleport" them away somewhere (because you can change their location) when you don't need them.

yuchen93, 2019-03-18:
Thanks for the immediate response. Can you give an example of how you would change the position of the objects? I have tried to do that by modifying "env.sim.data.body_xpos[env.cube_body_id]" in the Sawyer-Lift env but that does not seem to work for me.

yuchen93, 2019-03-25:
In case any body need this, I managed to change the pos with:
env.sim.model.geom_pos[env.sim.model.geom_name2id("cube")] = <new pose>

closed 2019-03-25 by yuchen93

kevin-thankyou-lin, 2022-05-03:
Thanks, Yuchen! In case it's helpful for anyone else, another way to change or set positions of objects is:

`env.sim.data.set_joint_qpos(<obj>, <pos_quat_array>)`

e.g. `env.sim.data.set_joint_qpos(env.model.mujoco_objects[0].joints[0], np.concatenate([np.ones(3)*j/100, np.array([1, 0, 0, 0])]))`