ARISE-Initiative/robosuite #24: what is "states(dataset) - flattened mujoco states"?

isk03276, 2019-03-19:
Hi, I am wondering what the "flattened mujoco states" is in the structure of demonstrations.
Can you tell what kind of data is contained in "states"?

Thanks!

[image]

amandlek, 2019-04-30:
This is the low-level mujoco state, which is the minimal information needed to reconstruct the state of the simulator. This can be used to playback the demonstrations, as shown [here](https://github.com/StanfordVL/robosuite/blob/master/robosuite/scripts/playback_demonstrations_from_hdf5.py#L100), frame by frame.

closed 2019-04-30 by amandlek

Gaoee, 2020-04-01:
Is there any document introduces what kind of states are contained in "states"? Because I want to use the positions of objects in the demonstrations data.

amandlek, 2020-04-01:
Take a look at [this](https://github.com/StanfordVL/robosuite/blob/master/robosuite/scripts/playback_demonstrations_from_hdf5.py#L80). You can just set the state of the environment, then retrieve whatever quantities you would like from the simulator itself (or by calling `env._get_observation()`)

Gaoee, 2020-04-02:
OK, thank you. So, if I want to get more observations which are not contained in demonstrations, I need to playback the demonstrations and retrieve from the environment.