ARISE-Initiative/robosuite #25: Why I got some dark images?

kaixindelele, 2019-03-25:
Hi,
When I try to save some images I found there are some black in the normal images.
I want to know how can I avoid it and if I don't render the env, will it exist?
the dark images:
[image]
[image]

the normal image:
[image]

thanks a lot!

kaixindelele, 2019-03-25:
the code:
```
import numpy as np
import robosuite as suite
import cv2

if __name__ == "__main__":

    # get the list of all environments
    a = suite.environments
    b = suite.environments.ALL_ENVS
    envs = sorted(suite.environments.ALL_ENVS)

    # print info and select an environment
    print("Welcome to Surreal Robotics Suite v{}!".format(suite.__version__))
    print(suite.__logo__)
    print("Here is a list of environments in the suite:\n")

    for k, env in enumerate(envs):
        print("[{}] {}".format(k, env))
    print()
    try:
        s = input(
            "Choose an environment to run "
            + "(enter a number from 0 to {}): ".format(len(envs) - 1)
        )
        # parse input into a number within range
        k = min(max(int(s), 0), len(envs))
    except:
        print("Input is not valid. Use 0 by default.")
        k = 0

    # initialize the task
    env = suite.make(
        envs[k],
        has_renderer=True,
        ignore_done=True,
        use_camera_obs=True,
        render_visual_mesh=True,
        control_freq=10,
    )
    env.reset()
    env.viewer.set_camera(camera_id=0)

    # do visualization
    for i in range(100):
        action = np.random.randn(env.dof)
        obs, reward, done, _ = env.step(action)

        image = obs['image']
        image = cv2.flip(image, 0)
        fileName = 'image_'+str(i)+'.jpg'
        cv2.imwrite(fileName, image)

        env.render()

```

amandlek, 2019-04-30:
I just tried running your script on my Macbook, and it seems to work just fine. Can you give further details of what machine you're running on and how I might reproduce the error?

kaixindelele, 2019-04-30:
> I just tried running your script on my Macbook, and it seems to work just fine. Can you give further details of what machine you're running on and how I might reproduce the error?

thanks.
I run this script in ubuntu16.04.
when I remove the 
        env.render()
I will work fine.

if I add it after env.step(action),  and save the image from obs["image"] can be randomly dark.

Now, when I found a way to avoid it just remove env.render(), so it may be not necessary rush to deal with it~

amandlek, 2019-04-30:
Ah I see - we've also seen similar problems on Ubuntu when trying to render on screen and off screen simultaneously. This could be a problem with `mujoco-py`, we're not positive.

kaixindelele, 2019-04-30:
may be a problem with mujoco-py and existed in ubuntu~
just remove env.render() may work well~

closed 2019-04-30 by kaixindelele