ARISE-Initiative/robosuite #35: Possible bug in UniformRandomSampler

eugval, 2019-07-02:
Hello,

That repository is fantastic, thank you for the good work!

I have noticed in the UniformRandomSampler under models.tasks.placementsampler that the z_rotation argument default is `"random"`, in the SaywerLift environment it is passed as `True` , and in the `sample_quat` function where it is used it seems that the only supported options are `[None, Iterable,  scalar]`.

In case I am misunderstanding something, and because I am not sure what the intended defaults were originally,   I am just raising an issue here for now!

Thanks!

kaixindelele, 2019-07-15:
> Hello,
>
> That repository is fantastic, thank you for the good work!
>
> I have noticed in the UniformRandomSampler under models.tasks.placementsampler that the z_rotation argument default is `"random"`, [...]

when I debug this value,I found the quat is the same every time.

amandlek, 2019-08-02:
Thanks for letting us know! You are both correct - this is a bug. Currently, passing the True value makes the sampler interpret the input as a fixed z-rotation of 1 radian. We will fix this soon, but the simple work around if you want uniformly sampled rotations is to just pass a value of None for that parameter.

closed 2020-03-09 by amandlek