ARISE-Initiative/robosuite #36: position of object does not change when I follow 'How  to build a custom environment'

yjc765, 2019-07-09:
Hello, thank you very much for the repo.
When I followed the steps of 'How to build a custom environment', I found a problem. In the 'adding the object' part, the example code is 
```
from robosuite.models.objects import BoxObject
from robosuite.utils.mjcf_utils import new_joint

object_mjcf = BoxObject()
world.merge_asset(object_mjcf)

obj = object_mjcf.get_collision(name="box_object", site=True)
obj.append(new_joint(name="box_object", type="free"))
obj.set("pos", [0, 0, 0.5])
world.worldbody.append(obj)
```

At this part I tried to change the position of the object, i.e change `obj.set("pos", [0, 0, 0.5])` into other numbers, but the rendering result shows the position of the object remains unchanged (at the bottom of the table, shown in the picture attached).
Has anybody also faced this problem and could you please give me some advice? Thank you very much!
[image]

yjc765, 2019-07-09:
OK, I found the solution. It should be like this `obj.set("pos", '0 0 0.5')`
Am I right?

yukezhu, 2020-02-08:
@yjc765 yes, you should set the string values for mjcf

closed 2020-02-08 by yukezhu