ARISE-Initiative/robosuite #37: Replay demos using action doesn't work

normandipalo, 2019-07-27:
I'm trying to replay some demos in various tasks and environments. While they work setting use-actions to False, they always fail setting it to True.
Is there a solution? Is it caused by some source of non-determinism?

amandlek, 2019-08-02:
Are you sure you're on the latest commit? I just verified that it works for me by collecting a demonstration and playing it back using the actions.

closed 2019-08-02 by amandlek

normandipalo, 2019-08-03:
I’m sorry, I omitted a detail: I’m referring to the set of demonstrations included in the repo at this link http://cvgl.stanford.edu/projects/roboturk/RoboTurkPilot.zip. 
Where they collected using a different implementation?
I wanted to use those since they are in great quantity instead of re-collecting demos.

amandlek, 2019-08-04:
Unfortunately, those demonstrations were collected before we had the functionality for deterministic action replay, so exact playback using the collected actions will not work.

YeeCY, 2020-07-28:
Which version of this repo should I use if I want to replay demonstrations in the RoboTurk dataset via action control?

amandlek, 2020-07-28:
As mentioned in the comments above, replaying the actions directly will not work, but the data itself can still be used for both online and offline policy learning - the [original paper](https://arxiv.org/abs/1811.02790) uses the demonstration data as resets for doing reinforcement learning (e.g. using [this](https://github.com/StanfordVL/robosuite/blob/master/robosuite/wrappers/demo_sampler_wrapper.py)) and [this paper](https://arxiv.org/abs/1911.05321) learns offline using the actions from the data. You should be on the master branch in order to make sure environment dynamics are consistent with the dataset.

YeeCY, 2020-07-28:
> As mentioned in the comments above, replaying the actions directly will not work, but the data itself can still be used for both online and offline policy learning - the [original [...]

Thanks, got it.