ARISE-Initiative/robosuite #40: Offscreen render appear black images(zeros array) as the increases of episode

kaixindelele, 2019-08-06:
As the episode increases, at the fixed step, obs will return a black images from obs until the end of the entire episode.

Very strangely, there are no problems with more than a hundred episodes at the beginning. The resulting images are normal and clearly.
Then once the bad picture appears, it will always appear regularly!

in [https://github.com/SurrealAI/surreal/blob/da705c02a243dbc7709c6002a02f1f8df6007674/surreal/main/ddpg_configs.py#L133](url)
I found that your team have also encountered similar bad pictures, I really want to know how you avoided it.

` env = suite.make(
        'SawyerLift',
        has_renderer=True,
        use_camera_obs=True,
        camera_depth=False,
        ignore_done=False,
        render_visual_mesh=False,
        reward_shaping=True,
        camera_height=Origin_size,
        camera_width=Origin_size,
        camera_name=camera_name,
        control_freq=10,
        reach_flag=False,
    )`

kaixindelele, 2019-08-07:
get bad images in fixed step may due to the DDPG.
Is it possible that some of the Robosuite's internal limitations trigger conditions that render to 0?

kaixindelele, 2019-08-24:
more details in here:
[https://github.com/openai/mujoco-py/issues/450](url)

csufangyu, 2019-10-21:
I meet the same error!Have you fixed it?

kaixindelele, 2019-10-21:
No, I just try to avoid it, what's your control algorithm? DDPG?
I guess there's a conflict between rendering time and network updates, though I don't have any way

csufangyu, 2019-10-21:
I don't use any control algorithm,I just want to known what does include in Observation,then I print it,and I find they are zeros.

kaixindelele, 2019-10-21:
不是吧，简简单单直接输出，应该不会出现黑图的。我测试过多次。。。

csufangyu, 2019-10-21:
robot-state [-0.60728109 -0.81142263  0.57186204  0.90653607 -0.99485061  0.55919608
  0.76308579  0.79448705  0.58445984  0.82034981 -0.42212837  0.10135222
  0.82903543  0.6462972   0.07870596 -0.02693952 -0.57281008 -0.30392208
  1.50980105 -0.02404782 -1.70234277 -0.00181675  0.0019294   0.57572215
 -0.1504085   0.90064105  0.79609005 -0.54423938  0.26242865  0.03428329]
image [[[0 0 0]
  [0 0 0]
  [0 0 0]
  ...
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ...
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ...
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 ...

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ...
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ...
  [0 0 0]
  [0 0 0]
  [0 0 0]]

 [[0 0 0]
  [0 0 0]
  [0 0 0]
  ...
  [0 0 0]
  [0 0 0]
  [0 0 0]]]
cube_pos [0.58169861 0.0155049  0.82143593]
cube_quat [0.         0.         0.47942554 0.87758256]
gripper_to_cube [-0.00597645 -0.1659134   0.07920512]
object-state [ 0.58169861  0.0155049   0.82143593  0.          0.          0.47942554
  0.87758256 -0.00597645 -0.1659134   0.07920512]

kaixindelele, 2019-10-21:
那你这是坏掉了啊，每次都这样？把代码贴出来看看

csufangyu, 2019-10-21:
我直接在demo.py里面改的！
`import numpy as np
import robosuite as suite

if __name__ == "__main__":

    # get the list of all environments
    envs = sorted(suite.environments.ALL_ENVS)

    # print info and select an environment
    print("Welcome to Surreal Robotics Suite v{}!".format(suite.__version__))
    print(suite.__logo__)
    print("Here is a list of environments in the suite:\n")

    for k, env in enumerate(envs):
        print("[{}] {}".format(k, env))
    print()
    try:
        s = input(
            "Choose an environment to run "
            + "(enter a number from 0 to {}): ".format(len(envs) - 1)
        )
        # parse input into a number within range
        k = min(max(int(s), 0), len(envs))
    except:
        print("Input is not valid. Use 0 by default.")
        k = 0

    # initialize the task
    env = suite.make(
        envs[k],
        has_renderer=True,
        ignore_done=True,
        use_camera_obs=True,
        control_freq=100,
    )
    env.reset()
    env.viewer.set_camera(camera_id=0)

    # do visualization
    for i in range(1000):
        action = np.random.randn(env.dof)
        obs, reward, done, _ = env.step(action)
        print("obs")
        for name, value in obs.items():
            print(name, value)
        print("reward is:", reward)
        print("action is:", done)
        print("something else")
        print(len(_))
        for name, value in _.items():
            print(name, value)
        # print("something else is", _.size())
        env.render()`

kaixindelele, 2019-10-21:
别渲染了，
has_render=False!

csufangyu, 2019-10-21:
不渲染是ok的，感谢！

kaixindelele, 2019-10-21:
有机会帮我测试一下，跑跑控制算法，会不会黑图。。。

csufangyu, 2019-10-21:
好的！你用DDPG控制起来效果咋样？

kaixindelele, 2019-10-21:
一言难尽，效果不好

csufangyu, 2019-10-21:
回报过于稀疏了，绝大部分都是0

kaixindelele, 2019-10-21:
可以设置为dense的，lift那个task有的

csufangyu, 2019-10-21:
好的，我试一试！可以加个微信吗？-------- 原始邮件 --------主题：Re: [StanfordVL/robosuite] Offscreen render appear black images(zeros array) as the increases of episode (#40)发件人：kaixindelele 收件人：StanfordVL/robosuite 抄送：universe ,Comment 可以设置为dense的，lift那个task有的

kaixindelele, 2019-10-21:
可以加这个Q群799378128

closed 2020-06-01 by yukezhu