ARISE-Initiative/robosuite #42: Memory leak of the IK controller?

eugval, 2019-08-16:
Hello!

I would like to use the ``ik_wrapper'' and the ``sawyer_ik_controller'' in order to control the robot in end-effector pose, but I am noticing that my environment consumes more and more memory over time!

Because I want to use distributed environments in order to  speed up learning,  my program ends up running out of memory.

I am wandering if this could be a memory leak somewhere in the controller? The only thing that I could think of is pybullet (looking at the code, i don't see where else this could be coming from).

Any Ideas? Perhaps any recommendation on what other packages I could use to replace pybullet do the the inverse kynematics?

Many Thanks,
Eugene

amandlek, 2019-08-17:
[Here's](https://github.com/StanfordVL/bullet3_ik) a fix that you can use - you'll need to compile it from source.

closed 2019-08-17 by amandlek

eugval, 2019-08-19:
Great! Thank you, that worked!!