ARISE-Initiative/robosuite #52: Problem playing back demonstrations using inverse kinematics

febert, 2020-01-15:
Hi, 

I'm trying to play back demonstrations using inverse kinematics using this script:
https://github.com/kpertsch/robosuite/blob/dev_frederik/robosuite/scripts/replay_dataset.py#L60

Using the "bins-Bread" data it produces motions that look random and do not pick up anything.

Could this be an issue with the control_freq parameter? Was this data collected with control_freq=100?

Thanks at lot!

amandlek, 2020-01-16:
Is this using RoboTurk data, or data that you collected yourself? The RoboTurk dataset doesn't support open-loop deterministic action playback unfortunately (the fix for this wasn't implemented in Robosuite until after collecting the data).

closed 2020-01-16 by amandlek

febert, 2020-01-16:
I was trying to use Roboturk data. What was the fix to it?

amandlek, 2020-01-16:
https://github.com/StanfordVL/robosuite/blob/master/robosuite/scripts/collect_human_demonstrations.py#L70