ARISE-Initiative/robosuite #53: Computing inversere kinematics for absolute endeffector position

febert, 2020-01-20:
Hi,

I'd like to compute inverse kinematics for an absolute endeffector position. I'm trying to reuse parts of RoboSuite's inverse kinematics controller for this.

Here is my attempt at doing it. 
https://github.com/kpertsch/robosuite/blob/dev_frederik/robosuite/scripts/test_absolute_inv_kin.py
Unfortunately the solution is incorrect, am I missing some tranformations? (I added the transform from base to world but that didn't help)

Thanks!

amandlek, 2020-01-21:
The only thing that stuck out to me is [this line](https://github.com/kpertsch/robosuite/blob/dev_frederik/robosuite/scripts/test_absolute_inv_kin.py#L10). You need to pass a function here that can be used to query for the robot's current joint positions. Currently, it only returns a static set of joint positions, so the joint controller in the SawyerIKController is not running closed-loop. Hope that helps!

closed 2020-01-21 by amandlek

josyulakrishna, 2023-05-17:
> https://github.com/kpertsch/robosuite/blob/dev_frederik/robosuite/scripts/test_absolute_inv_kin.py

Hi @febert  I'm trying to accomplish the same, this code has been deleted now, do you have any copy of it and if yes could you please share? thank you!