ARISE-Initiative/robosuite #56: Off-screen render images are incorrectly mirrored

SudeepDasari, 2020-01-28:
I've found that images rendered off-screen by mujoco_py (w/ `sim.render`) are upside down and mirrored when compared to images rendered by the Mujoco viewer. While the former issue can be fixed by adjusting the camera transform, the later issue can only be fixed by flipping the image buffers in code.

For example, below is the default image returned by the `sawyer_pick_place` environment, a rotated version of it for convenience, and the image shown by the Mujoco Viewer. Note that the bin is mirrored between the rotated and "real" images. 

I believe the Mujoco viewer's rendering is correct, as the right baxter gripper appears on the **robot's left** in the sim rendering. To fix this issue I flipped the image buffer's width channel in code, though there may be a better way to resolve this.

| From sim | Rotate 180 degrees | From viewer |
| --- | --- | --- |
| [image] | [image] | [image]

amandlek, 2020-03-03:
This is indeed true, and an issue with either mujoco-py or mujoco itself. For most use cases - I've been manually flipping the image axis myself, as you have done - since we rely on the `env.sim.render` call, which comes directly from mujoco-py.

closed 2020-03-03 by amandlek

IcarusWizard, 2022-12-08:
Hey guys. I just encounter the same issue in the latest version. Is there any plan to fix the bug within this repo? Or do we have to do the flipping manually?

amandlek, 2022-12-08:
You can just set [this option](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/macros.py#L28) appropriately to handle the flipping for you