ARISE-Initiative/robosuite #69: Playback Demostrations Failed

jiaqixuac, 2020-03-30:
I want to replay the demonstrations.

However, when I "python playback_demonstrations_from_hdf5.py --folder ../models/assets/demonstrations/SawyerPickPlace/ --use-actions" under the robosuite/robosuite/scripts/, the program failed the assertion and the robot can't grab the things.
[image]
[image]

I failed when installing mujoco-py==2.0.2.2 & robosuite==0.3.0 as well as mujoco-py==1.50.1.68 & robosuite==0.2.0.
Could you give any suggestion？

BTW, when I installed surreal, it says 
    "ERROR: surreal 0.2.1 has requirement mujoco-py<1.50.2,>=1.50.1, but you'll have mujoco-py 2.0.2.2 which is incompatible."
    "ERROR: robosuite 0.3.0 has requirement mujoco-py==2.0.2.2, but you'll have mujoco-py 1.50.1.68 which is incompatible."
Does this mean RoboTurk collected demonstrations using mujoco-py 1.50.1.68 and maybe incompatible with the current robosuite?

jiaqixuac, 2020-03-30:
I see the previous issue https://github.com/StanfordVL/robosuite/issues/37

So, it means the RoboTurk dataset does not promise replay using actions, and we can't use the RoboTurk dataset for some imitation learning methods like simple behavior cloning?

Thanks!

amandlek, 2020-04-02:
Just because the replay fails doesn't mean that you can't use it for simple behavior cloning, or other imitation learning methods! For example, we used the RoboTurk dataset in [this work](https://sites.google.com/stanford.edu/iris/) with purely offline learning.

closed 2020-04-02 by amandlek

xtkisscs, 2021-02-19:
` if j < num_actions - 1:
                    env.sim.set_state_from_flattened(states[j+1])`
Hey i change this code snippet to above to set state rather than assert state,then robot can perform correctly:)