ARISE-Initiative/robosuite #88: Panda IK Controller --> v.1.0

guyjacob, 2020-05-04:
First of all, many thanks for this project. The integration of Panda and Mujoco is a particular godsend :).

I noticed that as part of the major overhaul you guys are doing as part of the v1.0 branch, the way inverse kinematics controllers work changed.
My question - is this purely an API change/refactoring, or is the functionality improved as well? As you probably realized by now, I'm specifically wondering about IK control for the Panda robot.
If indeed this functionality is improved in v1.0, how safe/stable is it to use at the moment? Do you have any estimate when it might be released?

gal-leibovich, 2020-05-26:
Hi Guys,

Thanks a lot for this awesome project.
Adding to the question listed above, is there a planned target date/month/quarter for the 1.0 release?

cremebrule, 2020-05-28:
Sorry for getting to this so late -- the bulk of the IK changes have been refactoring, though we have extended functionality to include a linear interpolator, if so desired. The vanilla IK controller should be stable to use, though we're still ironing out the interpolator.

We hope to have v1.0 released by summer!

I'm closing this issue now, but feel free to open a new issue if you run into any problems with the IK controller.

closed 2020-05-28 by cremebrule