ARISE-Initiative/robosuite #94: Error occurred when running gripper selection demo

yukezhu, 2020-06-11:
Command: `python demo_gripper_selection.py`

```bash
/Users/yukez/Research/anaconda2/envs/robosuite/lib/python3.7/site-packages/gym/logger.py:30: UserWarning: WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
Creating window glfw
Traceback (most recent call last):
  File "demo_gripper_selection.py", line 27, in <module>
    control_freq=100,  # control should happen fast enough so that simulation looks smooth
  File "/Users/yukez/RoboticsSuite/robosuite/environments/base.py", line 22, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/Users/yukez/RoboticsSuite/robosuite/environments/pick_place.py", line 218, in __init__
    camera_depths=camera_depths,
  File "/Users/yukez/RoboticsSuite/robosuite/environments/robot_env.py", line 180, in __init__
    hard_reset=hard_reset,
  File "/Users/yukez/RoboticsSuite/robosuite/environments/base.py", line 109, in __init__
    self._reset_internal()
  File "/Users/yukez/RoboticsSuite/robosuite/environments/pick_place.py", line 479, in _reset_internal
    super()._reset_internal()
  File "/Users/yukez/RoboticsSuite/robosuite/environments/robot_env.py", line 259, in _reset_internal
    super()._reset_internal()
  File "/Users/yukez/RoboticsSuite/robosuite/environments/base.py", line 199, in _reset_internal
    self._get_reference()
  File "/Users/yukez/RoboticsSuite/robosuite/environments/pick_place.py", line 445, in _get_reference
    self.sim.model.geom_name2id(x) for x in self.robots[0].gripper.right_finger_geoms
  File "/Users/yukez/RoboticsSuite/robosuite/environments/pick_place.py", line 445, in <listcomp>
    self.sim.model.geom_name2id(x) for x in self.robots[0].gripper.right_finger_geoms
  File "mujoco_py/generated/wrappers.pxi", line 1035, in mujoco_py.cymj.PyMjModel.geom_name2id
ValueError: No "geom" with name r_finger exists. Available "geom" names = ('floor', 'robot0_controller_box', 'robot0_pedestal_feet', 'robot0_torso', 'robot0_pedestal_vis1', 'robot0_pedestal_col1', 'robot0_base_vis', 'robot0_base_col', 'robot0_link0_visual', 'robot0_link0_collision', 'robot0_head_visual', 'robot0_head_collision', 'robot0_screen_visual', 'robot0_screen_collision', 'robot0_link1_visual', 'robot0_link1_collision', 'robot0_link2_visual', 'robot0_link2_collision', 'robot0_link3_visual', 'robot0_link3_collision', 'robot0_link4_visual', 'robot0_link4_collision', 'robot0_link5_visual', 'robot0_link5_collision', 'robot0_link6_visual', 'robot0_link6_collision', 'gripper0_l_finger', 'gripper0_l_finger_visual', 'gripper0_l_finger_tip', 'gripper0_l_finger_tip_visual', 'gripper0_r_finger', 'gripper0_r_finger_visual', 'gripper0_r_finger_tip', 'gripper0_r_finger_tip_visual', 'robot0_right_l4_2', 'robot0_right_l2_2', 'robot0_right_l1_2', 'Milk0', 'Bread0', 'Cereal0', 'Can0').```

yukezhu, 2020-06-11:
Looks like it is looking for `gripper0_r_finger`

cremebrule, 2020-06-12:
fixed!

closed 2020-06-12 by cremebrule