ARISE-Initiative/robosuite #97: Error raise up when run the "playback_demonstrations_from_hdf5.py" with option "--use-actions"

Fitz13009, 2020-06-16:
Hello,

I am a PhD student working in the robotic imitation learning area.
Recently I am trying to run the script "playback_demonstrations_from_hdf5.py" to reply the human's demonstration from RoboTurk dataset. However, when I add the option "--use-actions" to argument list, an exception raised up, which complains an out of bounds error for the index. 

The bug comes from the line 560 of the script "sawyer_pick_place.py", where the code asks for the 6th to 9th elements of a list containing only 6 elements.

I will appreciate any help or suggestions from you.

Thank you in advance.

All the best,
Ge Li

Fitz13009, 2020-06-16:
My temp solution is to avoid calling function "self._gripper_visualization()", but the human demostration replay will still break the assertion (line 96 in "playback_demonstrations_from_hdf5.py"), when comparing the state after the action has been executed with the state stored in the RoboTurk dataset. 

If I make this assertion a comment, the simulation will run properly but in most cases, the robot will fail in grasping the object.

Fitz13009, 2020-06-16:
I think issue #37 and #69 can explain the assertion error, but the error to index out of bound still remains...

amandlek, 2020-06-16:
Thanks for bringing up this issue. Can you provide more details, such as which exact dataset you were trying to playback?

Fitz13009, 2020-06-17:
> Thanks for bringing up this issue. Can you provide more details, such as which exact dataset you were trying to playback?

Hi, thank you for your reply. After my testing, all the datasets with "bins" raise the error to the "out of bound in index", but the "pegs" are fine. 

[image]

And all of them will break the assertation when comparing states.

jaustinb1, 2020-06-17:
Hi @Fitz13009 thanks for bringing this issue up. This issue exists as the xml used for the robot has changed since we collected the data. There are two possible solutions here 
1. you can make an environment by passing gripper_visualization=False when making the environment 
2. you can comment out the 4 sites created https://github.com/StanfordVL/robosuite/blob/master/robosuite/models/assets/robots/sawyer/robot.xml#L120

We are getting ready to make a robosuite v1.0 release where this issue will be fixed.

closed 2020-06-17 by jaustinb1

Fitz13009, 2020-06-19:
> Hi @Fitz13009 thanks for bringing this issue up. This issue exists as the xml used for the robot has changed since we collected the data. There are two possible solutions here
>
> 1. you can make an [...]

Hi,

thank you very much. I am looking forward to seeing your 1.0 release...

All the best,
Ge Li