ARISE-Initiative/robosuite #98: demo_gripper_interaction does not work

kciebiera, 2020-06-23:
API changed:

```
krzysztof@krzysztof-MS-7982:~/tmp/tst/robosuite/robosuite$ python scripts/demo_gripper_interaction.py 
Traceback (most recent call last):
  File "scripts/demo_gripper_interaction.py", line 17, in <module>
    from robosuite.models.gripper import TwoFingerGripper
ModuleNotFoundError: No module named 'robosuite.models.gripper'
```

kciebiera, 2020-06-23:
here goes the diff:
```
diff --git a/robosuite/scripts/demo_gripper_interaction.py b/robosuite/scripts/demo_gripper_interaction.py
index 34c5a3d..0436b2e 100644
--- a/robosuite/scripts/demo_gripper_interaction.py
+++ b/robosuite/scripts/demo_gripper_interaction.py
@@ -14,8 +14,8 @@ from mujoco_py import load_model_from_path, MjSim, MjViewer
 
 from robosuite.models import MujocoWorldBase
 from robosuite.models.arenas.table_arena import TableArena
-from robosuite.models.gripper import TwoFingerGripper
-from robosuite.models.mujoco_object import BoxObject
+from robosuite.models.grippers import TwoFingerGripper
+from robosuite.models.objects.generated_objects import BoxObject
 from robosuite.utils.mjcf_utils import new_joint, new_actuator
```

yukezhu, 2020-09-28:
Closed as issue fixed in our v1.0 release

closed 2020-09-28 by yukezhu