ARISE-Initiative/robosuite #104: How to add equality constraints between an object and the robot gripper dynamically?

prasoongoyal, 2020-09-06:
I'm trying to simulate a suction-type gripper, i.e., when the robot's gripper is sufficiently close to an object, the object is attached to the gripper. I suppose I can use [weld equality constraint](http://www.mujoco.org/book/XMLreference.html#equality-weld), but I'm not sure how to add it dynamically. I tried adding the following code to the `__init__` method of the environment class:
```
elem = ET.Element('weld')
elem.set('body1', 'right_gripper')
elem.set('body2', 'Can0')
self.model.equality.append(elem)
```
but the `env.sim.model.eq_type` is blank after the environment is created.

Not sure how to fix this.

prasoongoyal, 2020-09-14:
For anyone who runs into this issue -- one way is to add the constraint to the XML object that is used to create the MjModel.

closed 2020-09-14 by prasoongoyal

Gaoee, 2020-11-06:
Hi, I also want to use a suction-type gripper. Are you successful in using such gripper in your task?