ARISE-Initiative/robosuite #107: [Question] Custom robot?

chisarie, 2020-09-29:
Hi, thank you very much for the nice work! 
I am going to start a project involving mobile manipulation, i.e. a robot arm on top of a wheeled platform.
Do you think it is possible for me to add the custom robot model in your robosuite? If yes, do you have any guidelines? 
If I manage to get started with this, I will also be happy to contribute the new robot model and task to the community.

Thanks

yukezhu, 2020-10-02:
Thank you for reaching out to us. We sincerely welcome community contributions. A contribution guideline is coming up soon. Mobile manipulators would be a potential extension to the current version of robosuite. You can start this in a fork of our repo, and make pull requests when it is completed. Thanks!

chisarie, 2020-10-05:
Thank you very much for the answer! If I will start with robosuite, I will definitely let you know once I have interesting features to share.
Thanks again and keep up with great work :)

closed 2020-10-05 by chisarie