ARISE-Initiative/robosuite #108: Blank screen

varun-intel, 2020-09-30:
I'm trying to create a new environment by modifying your PickPlace example and am running into a strange error. In order to simplify the environment, I'd like to remove the objects placed in the bins. However, reducing the number of objects down to one starts causing issues, producing a blank screen after a couple hundred steps.

I'm attaching a minimal example. The example should run without a problem as is; you can reproduce the error by commenting lines 217 - 225 and uncommenting 227 - 234.

https://gist.github.com/varun-intel/b3a75d9a579967014c2cdd332fb9ce8d

Do you have any suggestions for debugging? It would be helpful if I could have the library print out some more logs.

cremebrule, 2020-10-01:
This sounds like it's related to a problem we've had for awhile; specifically, black screens being incorrectly rendered either during resets or (as you mention) mid-sim. It's an inherent problem with the MjViewer, and unfortunately not something we can directly fix as that problem is tied to the underlying mujoco engine.

A couple workarounds:
1. You can call an extra `env.reset()` each time you start a new episode; we've found this to reduce these occurances
2. Use your environments with `hard_reset=False`. Most of our problems with this issue are tied to the destruction / re-initialization of the MjViewer, which only occurs with hard resets enabled.

P.S.: We already have OTS versions of the Single-object variations of PickPlace -- see [pick_place.py] for details!(https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/pick_place.py#L789). Just call the single-object version as the env name during the `robosuite.make()` env creation call. (: 

Hope that helps!

varun-intel, 2020-10-01:
Thanks for the prompt reply. I've seen the single-object options; this was more of an exercise for me to understand how the envs work.

hard_reset=False seems to fix the problem pretty reliably.

closed 2020-10-01 by varun-intel