ARISE-Initiative/robosuite #113: How to register a new env to robosuite?

HeegerGao, 2020-10-05:
Hi there, I want to build a new env and register it into robosuite so I can start my env like `env = suite.make(env_name='mynewenv', ……)`. But I don't find where you register all your initial tasks and environments. I just find a dictionary called `REGISTERED_ENVS` in `/robosuite/environments/base.py`. It's a little strange to me that I don't find where do you change the value of this dictionary's value. `REGISTERED_ENVS` equals to {} at the beginning of the code, but it seems full of envs automatically. Could you tell me how do you do that?

yukezhu, 2020-10-05:
The environment registration is done through the meta class: https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/base.py#L40

cremebrule, 2020-10-05:
Specifically, when you create a new environment class and want to register it so you can call it with its string name with `suite.make()`, simply import your custom environment class in the `__init__.py` [HERE](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/__init__.py).

Closing this issue for now; feel free to re-open it if you continue to have issues!

closed 2020-10-05 by cremebrule