ARISE-Initiative/robosuite #114: [Question] Off-screen rendering results in `ERROR: GLEW initalization error: Missing GL version`

hermanjakobsen, 2020-10-06:
For instance, creating the environment
```
env = robosuite.make(
    'Lift',
    robots='Panda',
    gripper_types="PandaGripper",
)
``` 
gives the mentioned error. The environment renders fine, however, using on-screen rendering
```
env_test = suite.make(
    'Lift',
    robots='Panda',
    gripper_types="PandaGripper",
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
    use_object_obs=False
)
```
I have checked out the [issues](https://github.com/openai/mujoco-py/issues/408) in the mujoco-py repo and it looks like this is a known problem with mujoco-py.
My question is: Do you guys know of any workaround to make off-screen rendering work? I have tried the tips given in the linked issue without any luck :(

yukezhu, 2020-10-06:
Hi @hermanjakobsen can you let us know your system runtime, like your OS and Python version? Thanks.

hermanjakobsen, 2020-10-06:
OS: Ubuntu 20.04.1 LTS
Python version: Python 3.8.2
I tried to use [pdb](https://docs.python.org/3/library/pdb.html) to debug, but I could not find any useful information from the stacktrace. 
If it is of any interest, my `LD_LIBRARY_PATH` is set to the following in `~/.bashrc`
```
export LD_LIBRARY_PATH=$WEBOTS_HOME/lib  # Add the Webots libraries to the library path (useful when launching Webots directly without using the launcher).
export LD_LIBRARY_PATH=/usr/lib/cuda/lib64:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/usr/lib/cuda/include:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/hermankj/.mujoco/mujoco200/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/hermankj/.mujoco/mjpro150/bin
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libGLEW.so:/usr/lib/x86_64-linux-gnu/libGL.so
```

cremebrule, 2020-10-07:
I don't think I've run into this issue before. Have you tried updating some of the packages, as suggested in this comment [here](https://github.com/openai/mujoco-py/issues/268#issuecomment-595177264)?

hermanjakobsen, 2020-10-07:
Yes, I have tried it without any luck ://

cremebrule, 2020-10-08:
Hmm, unfortunately none of our team is directly able to replicate the issue -- Ubuntu 20.04 is relatively new, and we officially support 16.04 / 18.04, and MacOS since that is what our team uses. Is there any way you might be able to test on an earlier distro of Ubuntu? I would also suggest posting this issue on the mujoco-py repo as it sounds like an issue directly related to mujoco.

hermanjakobsen, 2020-10-09:
I will try to look into it! Thank you for your help and time :)

closed 2020-10-10 by yukezhu

HeegerGao, 2020-10-17:
Hi, I have come to the same issue. When I use on_screen renderer, it works normally:
```
suite.make(
    **config,
    has_renderer=True,
    reward_shaping=True,
    render_camera='agentview',
    has_offscreen_renderer=False,
    use_camera_obs=False,
    hard_reset=False
    )
```
But when I use off_screen renderer, it shows 'ERROR: GLEW initalization error: Missing GL version':
```
suite.make(
    **env_config,
    has_renderer=False,  # no on-screen renderer
    has_offscreen_renderer=True,  # off-screen renderer is required for camera observations
    ignore_done=True,  # (optional) never terminates episode
    use_camera_obs=True,  # use camera observations
    camera_heights=84,  # set camera height
    camera_widths=84,  # set camera width
    camera_names="agentview",  # use "agentview" camera
    use_object_obs=False,  # no object feature when training on pixels
    reward_shaping=True,  # (optional) using a shaping reward
)
```
Do you have any ideas?

hermanjakobsen, 2020-10-17:
@HeegerGao, could you state your system runtime? It might be useful information for @cremebrule.

HeegerGao, 2020-10-17:
OK. I am using Ubuntu 18.04. The error message is like this:
```
ERROR: GLEW initalization error: Missing GL version

Press Enter to exit ...
```

HeegerGao, 2020-10-17:
And I am using mujoco200

HeegerGao, 2020-10-17:
Hi, I think I have solved this problem, though I am not quite sure about the principle behind it.

I found the same [issue](https://github.com/openai/mujoco-py/issues/408) in mujoco-py. They said we should do as following:
```
In Ubuntu 18.04.3 LTS:
sudo mkdir -p /usr/lib/nvidia-000
Then add this in ~/.bashrc file:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia-000
```
The I run my code again. It starts compiling mujoco_py/cymj.pyx and cythonizing mujoco_py/cymj.pyx, and after a while, it starts running normally.

According to the discussion in the [issue](https://github.com/openai/mujoco-py/issues/408), the problem may come from a too higher version of OpenGL.

That's my solution, and I would appreciate it if you can explain the principle behind it. Thank you!

cremebrule, 2020-10-18:
Hi HeegerGao,

Unfortunately, we don't have much control over the mujoco-py backend -- we've always had issues with the renderer specifically as well, which are still unsolved to date. So this sounds like one of those issues you simply have to find a workaround for :/

We do have some alternative rendering options in the works, so I would stay tuned for some updates in the near future!

xu-george, 2021-03-31:
add the following lines on the base.py, could solve the problem for me 

            if self.sim._render_context_offscreen is None:
                from mujoco_py import GlfwContext
                GlfwContext(offscreen=True)
                render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
                self.sim.add_render_context(render_context)
            self.sim._render_context_offscreen.vopt.geomgroup[0] = (1 if self.r

kaixindelele, 2021-04-15:
> add the following lines on the base.py, could solve the problem for me
>
> ```
>         if self.sim._render_context_offscreen is None:
>             from mujoco_py import GlfwContext [...]

hi, the code is not complete~

xu-george, 2021-04-15:
> hi, the code is not complete~
compare with original code, just one line is added

sqdy2669, 2024-09-19:
> 在 base.py 上添加以下几行，可以解决我的问题
> 
> ```
>         if self.sim._render_context_offscreen is None:
>             from mujoco_py import GlfwContext
>             GlfwContext(offscreen=True)
>             render_context = MjRenderContextOffscreen(self.sim, device_id=self.render_gpu_device_id)
>             self.sim.add_render_context(render_context)
>         self.sim._render_context_offscreen.vopt.geomgroup[0] = (1 if self.r
> ```

It really does help！