ARISE-Initiative/robosuite #116: [Question] How to control robot joints to a desired position?

hermanjakobsen, 2020-10-13:
As to my understanding, the following code should move the robot joints to a desired position
```
import robosuite as suite
from robosuite import load_controller_config

controller_config = load_controller_config(default_controller='JOINT_POSITION')

env = suite.make(
            'Lift',
            robots='UR5e',
            controller_configs = controller_config,
            has_renderer = True,
            has_offscreen_renderer= False,
            use_camera_obs=False,
            use_object_obs=False,
            control_freq = 30,
            render_camera = None,      
        )

env.reset()

goal_joint_pos = [np.pi / 2, 0, 0, 0, 0, 0, 0]

# Run random policy
for _ in range(3000):
    if env.done:
        break
    env.render()
    
    env.step(goal_joint_pos)

# close window
env.close()
```
However, when I run the code, it looks like a joint velocity is applied to joint 1 effectively ignoring the desired joint position. In other words, the robot just starts spinning around joint 1. 

I have tried to take a deep-dive into the code, but I have clearly misunderstood something. Any help on how to control the joints to a desired position would be greatly appreciated! :)

cremebrule, 2020-10-13:
Hi Herman,

This is the correct behavior; the joint position controller accepts relative (delta) position commands instead of absolute commands. So sending a command of `np.pi/2, 0, 0, 0, 0, 0, 0]` will keep all joints except the first at their current position while trying to rotate the first joint at a rate of np.pi/2 per controller timestep (note that this value is clipped by the controller since the default max input values are -1/+1).

If you’d like the manually set the absolute joint positions, currently the only way to do that is to call `env.robots[x].set_joint_positions(...)`, which manually sets the joint positions of the specified robot in sim.

The joint position controller also has a way to interpret the input as absolute commands, though there’s currently no way to configure this from the controller config file. I will add this to my TODOs since this is a feature that should be there!

Thanks for bringing this up, closing the issue for now, but feel free to re-open / start a new issue if you continue to run into problems with this controller.

hermanjakobsen, 2020-10-14:
If anyone is interested in a "hacky" way to control joint positions before Josiah has added the feature, the following code will do the trick
```
import numpy as np
import robosuite as suite

def relative2absolute_joint_pos_commands(goal_joint_pos, robot, kp, kd):
    assert len(goal_joint_pos) == robot.dof

    action = [0 for _ in range(robot.dof)]
    curr_joint_pos = robot._joint_positions
    curr_joint_vel = robot._joint_velocities

    for i in range(robot.dof):
        action[i] = (goal_joint_pos[i] - curr_joint_pos[i]) * kp - curr_joint_vel[i] * kd
    
    return action

def robosuite_simulation_controller_test(env, sim_time):
    # Reset the env
    env.reset()

    robot = env.robots[0]

    kp = 2
    kd = 1.2

    for t in range(sim_time):
        if env.done:
            break
        env.render()
        
        action = relative2absolute_joint_pos_commands([np.pi / 2, 0, 0, 0, 0, 0], robot, kp, kd)

        if t > 1200:
            action = relative2absolute_joint_pos_commands([0, -np.pi/4, 0, 0, 0, 0], robot, kp, kd)
        elif t > 800:
            action = relative2absolute_joint_pos_commands([0, 0, 0, 0, 0, 0], robot, kp, kd)
        elif t > 400:
            action = relative2absolute_joint_pos_commands([3*np.pi/2, 0, 0, 0, 0, 0], robot, kp, kd)

        observation, reward, done, info = env.step(action)

    # close window
    env.close() 

env = suite.make(
            'Lift',
            robots='UR5e',
            gripper_types=None,
            has_renderer=True,
            has_offscreen_renderer=False,
            use_camera_obs=False,
            use_object_obs=False,
            control_freq=50,
            render_camera=None,
            horizon=2000,      
        )

robosuite_simulation_controller_test(env, env.horizon)
```
Note that the controller has not been tuned, thoroughly tested nor using smallest signed angles.