ARISE-Initiative/robosuite #117: how to

kaixindelele, 2020-10-13:
Can we cancel the creation of the glew window?
This window will pop up every round. It really affect the experiment!
In addition, when I run a experiment remotely in local, even if cancel render image, there will always be an error:
> mujoco_ py.cymj.GlfwError : Failed to initialize GLFW

cremebrule, 2020-10-13:
Hi kaixindelele,

You should be able to create an environment without any renderer. When you can call `suite.make(...)`, can you make sure you set `has_renderer=False` and `has_offscreen_renderer=False` (note that in order to do the latter you also have to set `use_camera_obs=False` as well)?

kaixindelele, 2020-10-14:
Yes, I am sure, I set has_`renderer=False` and has_`offscreen_renderer=False` and `use_camera_obs=False`
When I use gym fetch env, we can render a image without creating any glew window, why `Robosuite` must  use glew ?

cremebrule, 2020-10-18:
Sorry for the delay -- the window shouldn't be popping up if you haven't set any renderer to be True. What setup are you using?

kaixindelele, 2020-10-18:
It's very strange, different computers have different phenomena, when 
>GPU-1080
>Intel® Xeon(R) CPU E5-2650 v2 @ 2.60GHz × 32 
>llvmpipe (LLVM 9.0, 256 bits)
> mujoco-py                 2.0.2.13
it can render image without glew windows, with has_render=False, has_offscreen_renderer=True, use_camera_obs=False.

In another computer:
>GPU-2080Ti
>Intel(R) Xeon(R) Silver 4210 CPU @ 2.20GHz
>mujoco-py                 2.0.2.8

Is it related to computer hardware or mujoco version? 
Because it can also render image without glew windows on another machine using mujoco-py1. X

cremebrule, 2020-12-18:
Were you ever able to resolve this issue?

yukezhu, 2021-01-08:
Closed as no response from OP

closed 2021-01-08 by yukezhu

kaixindelele, 2021-03-19:
> Were you ever able to resolve this issue?

After testing, the same code, it have to create a glew window with offscreen on two different Ubuntu16.04 computers;
On the another two Ubuntu 18.04 computers, offscreen does not need to create a glew window;
I wonder if you also have this phenomenon?