ARISE-Initiative/robosuite #118: Adding new papers using robosuite

yukezhu, 2020-10-23:
"Variational Imitation Learning with Diverse-quality Demonstrations" Voot Tangkaratt, Bo Han, Mohammad Emtiyaz Khan, Masashi Sugiyama, ICML 2020. https://proceedings.icml.cc/static/paper_files/icml/2020/577-Paper.pdf

yukezhu, 2020-11-01:
"Conservative Safety Critics for Exploration" Homanga Bharadhwaj, Aviral Kumar, Nicholas Rhinehart, Sergey Levine, Florian Shkurti, Animesh Garg. https://arxiv.org/abs/2010.14497

yukezhu, 2020-11-16:
"Transformers for One-Shot Visual Imitation" Sudeep Dasari, Abhinav Gupta. https://corlconf.github.io/paper_463/

yukezhu, 2020-11-28:
"Deep Reinforcement Learning for Contact-Rich Skills Using Compliant Movement Primitives" Oren Spector, Miriam Zacksenhouse. https://arxiv.org/abs/2008.13223

cremebrule, 2020-12-18:
fixed in #137

closed 2020-12-18 by cremebrule