ARISE-Initiative/robosuite #119: Running `self._check_gripper_contact()` results in `AttributeError: 'SingleArm' object has no attribute 'arm_type'`

hermanjakobsen, 2020-10-26:
Hi! 
I have created my own environment and I am interested in checking whether there is contact between the robot's gripper and an object. However, when I run `self._check_gripper_contact()` I get the following error message `AttributeError: 'SingleArm' object has no attribute 'arm_type'. 
The method is called in the `_get_observation(self)` method:
``` 
def _get_observation(self):
        """
        Returns an OrderedDict containing observations [(name_string, np.array), ...].
        Important keys:
            `'robot-state'`: contains robot-centric information.
            `'object-state'`: requires @self.use_object_obs to be True. Contains object-centric information.
            `'image'`: requires @self.use_camera_obs to be True. Contains a rendered frame from the simulation.
            `'depth'`: requires @self.use_camera_obs and @self.camera_depth to be True.
            Contains a rendered depth map from the simulation
        Returns:
            OrderedDict: Observations from the environment
        """
        di = super()._get_observation()

        di['contact'] = self._check_gripper_contact()

        return di
```
I have tried to call the method with the UR5e robot using the PandaGripper.

yukezhu, 2020-11-01:
Hi @hermanjakobsen Were you able to resolve this issue yourself? The following code shows an example of how to create an env with UR5e robot and the Panda gripper. And you can access all contact information from `env.sim.data`. If would be useful if you provide more information/code snippet about your concrete problem.

```python
from robosuite.controllers import load_controller_config
from robosuite.utils.input_utils import *

if __name__ == "__main__":

    # Create dict to hold options that will be passed to env creation call
    options = {}

    # Choose environment and add it to options
    options["env_name"] = "Lift"
    options["robots"] = "UR5e"
    options["gripper_types"] = "PandaGripper"

    # Choose controller
    controller_name = "OSC_POSE"

    # Load the desired controller
    options["controller_configs"] = load_controller_config(default_controller=controller_name)

    # initialize the task
    env = suite.make(
        **options,
        has_renderer=True,
        has_offscreen_renderer=False,
        ignore_done=True,
        use_camera_obs=False,
        control_freq=20,
    )
    env.reset()
    env.viewer.set_camera(camera_id=0)

    # Get action limits
    low, high = env.action_spec

    # do visualization
    for i in range(10000):
        action = np.random.uniform(low, high)
        obs, reward, done, _ = env.step(action)
        env.render()
```

cremebrule, 2020-11-01:
Hi @hermanjakobsen , so for taking so long to get around to this. Great catch; that is a method that was referencing a slightly outdated version architecture -- we never caught this because that method wasn't explicitly used in any of our test / use cases!

If you want to directly fix this on your side immediately, just change `robot.arm_type` [HERE](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/robot_env.py#L430) to be `robot.robot_model.arm_type` instead.

I'll be running a few bug fixes this week, so I'll close this issue once the bug is actually fixed on the master branch. Huge thanks for bringing up these issues; this is really helping iron out some of the more niche use case bugs :)

hermanjakobsen, 2020-11-02:
Hi @yukezhu, I was able to resolve the issue. I just modified the `self._check_gripper_contact()` to 
```
def _check_gripper_contact(self):
        """
        Checks whether each gripper is in contact with an object.
        Returns:
            list of bool: True if the specific gripper is in contact with an object
        """
        collisions = [False] * self.num_robots
        for idx, robot in enumerate(self.robots):
            for contact in self.sim.data.contact[: self.sim.data.ncon]:
                if (
                    self.sim.model.geom_id2name(contact.geom1)
                    in robot.gripper.contact_geoms
                    or self.sim.model.geom_id2name(contact.geom2)
                    in robot.gripper.contact_geoms
                ):
                    collisions[idx] = True
                    break

        return collisions
```
This works for both the UR5e and Panda, but I haven't tested it for other robots (e.g two-armed robots). I think the fix proposed by @cremebrule is much better though 😅

cremebrule, 2020-12-18:
(implicitly) fixed in #137

closed 2020-12-18 by cremebrule