ARISE-Initiative/robosuite #120: [Question] How to change the orientation frame of a gripper?

hermanjakobsen, 2020-10-27:
Hi! 
I have tried to implement my own gripper. The gripper consists purely of a rigid body and is meant to simulate an ultrasound probe, and I have been able to implement the gripper with a pose that matches how it would be placed in the real world. However, the orientation frame of the gripper is somewhat wrong. I would like the z-axis to point out of the tip of the probe, as it is done with the already implemented grippers. Is it possible to change or redefine the orientation frame of a gripper? 

The xml-file for the gripper is 
```
<mujoco model="ultrasound_probe_gripper">
  <asset>
    <mesh file="meshes/ultrasound_probe/Probe_holder_C3_8_probe_clean.stl" name="probe_mesh" scale="0.001 0.001 0.001"/>
  </asset>
  <worldbody>
      <body name="gripper_base" pos="-0.004 -0.063 0.12" quat="0 0 0.7071 -0.7071">
        <!--site name="ft_frame" pos="0 0.0185 0" size="0.036 0.025 0.017" rgba="1 0 0 1" type="ellipsoid" group="1"/-->
        <!-- Standard site for robosuite below. -->
        <site name="ft_frame" size="0.01 0.01 0.01" rgba="1 0 0 1" type="sphere" group="1"/> 
        <geom name="probe_collision" mesh="probe_mesh" type="mesh" mass="1" group="0"/>
        <geom name="probe_visual" mesh="probe_mesh" type="mesh" group="1"/>
        <!-- This site was added for visualization. -->
        <site name="grip_site" pos="0 0 0.1050" size="0.01 0.01 0.01" rgba="1 0 0 1" type="sphere" group="1"/>
        <!-- This site was added for visualization. -->
        <site name="grip_site_cylinder" pos="0 0 0.1399" size="0.005 10" rgba="0 1 0 0.3" type="cylinder" group="1"/>
    </body>
  </worldbody>
  <sensor>
        <force name="force_ee" site="ft_frame"/>
        <torque name="torque_ee" site="ft_frame"/>
    </sensor>
</mujoco>
```

I have attached two images to show the setup.
[image]
[image]

yukezhu, 2020-11-01:
Yes, I believe you can do so by changing the `quat` value in `<body name="gripper_base" pos="-0.004 -0.063 0.12" quat="0 0 0.7071 -0.7071">`.

hermanjakobsen, 2020-11-02:
Changing the `quat` value will also rotate the probe, which results in a non-realistic setup (see attached images). The reason for me wanting to have the z-axis pointing out of the probe is because of the readings (i.e. force along each axis) from the force sensor being flipped when the frame is oriented as it is now. It would be nice if the force reading from the probe would match the force readings from one of the other grippers. 
Overall, this "issue" is not very important. :) 
[image]
[image]

yukezhu, 2020-11-02:
In this case, you might want to attach the site to another child element of the body instead, so the end-effector stays in the same place while you make changes to the frame. Alternatively, you can always do post-processing to change the frame.

closed 2020-11-02 by yukezhu

danielstankw, 2022-05-08:
@hermanjakobsen I know this issue is super old but I cant find any help for the issue I have. 
Do you know where can I set visibility of the reference frames of different sites/ bodies like you have? 
Mine are super thick and thus obscure the view.,