ARISE-Initiative/robosuite #122: Is it able to use a suction-type gripper?

Gaoee, 2020-11-06:
Hi, I am really interesting in this project. I want use a suction-type gripper in my task. Are there any suggestions you can give? Thanks.

cremebrule, 2020-11-07:
Hi @Gaoee ,

Yes, as another user mentioned in the thread you replied to ([#104 ](https://github.com/ARISE-Initiative/robosuite/issues/104#issuecomment-691788076)), using equality (weld) constraints would be a viable way to do this. The method he described should work, but unfortunately the sim object is already created by the time you can access `self.model`, so any further elements you try adding won't actually do anything.

There's unfortunately no clean and easy way to do this robustly for all objects, so you'll have to hard code stuff for now. As prasoongoyal mentioned, in any raw XML file that is used in your sim, you can directly add the correct `<weld ...>` tag in an `<equality><equality/>` section in that XML. See the [MujocoXML reference](http://www.mujoco.org/book/XMLreference.html#equality) for details. Set `active=false` by default, and then when a specific condition is met during runtime (e.g.: the suction gripper is contacting the object), you can set the weld to be active, thus simulating the "suctioning" effect

We also currently have some projects in the works that are also using weld constraints, so I'm thinking of porting some of this high-level functionality into robosuite for users such as yourself to use. So stay tuned!

Gaoee, 2020-11-12:
Really thanks for your replay.
I will try your suggestion on my task. But maybe it still not that easy to simulate pressure of a suction gripper.

closed 2020-11-12 by Gaoee