ARISE-Initiative/robosuite #128: Plans for providing forward/inverse Kinematics, access to the Jacobian??

TheCamusean, 2020-11-14:
Hello,

I was looking around the suite and I realize the environments don't provide essential robotics function such as Forward/Inverse Kinematics, Inverse Dynamics, Jacobian between configuration and task space.

Do you have any plan of integrating such features?

Best regards,
-Julen

cremebrule, 2020-11-19:
Hi Julen,

Could you perhaps be a bit more specific? We currently support IK control for our original robosuite v0.3 robots (Panda, Sawyer, and Baxter), and Jacobians for any arbitrary mujoco body / geom / site is already supported by mujoco-py (see their [documentation](https://openai.github.io/mujoco-py/build/html/reference.html) for method calls).

cremebrule, 2020-12-18:
As this is a stale issue, I'm going to close this for now. Feel free to re-open it if you have any updates!

closed 2020-12-18 by cremebrule