ARISE-Initiative/robosuite #134: How to adjust size and position of the object loaded by xml file?

HankerSia, 2020-12-03:
I load a object using the following code:
from robosuite.models.objects.objects import MujocoXMLObject
seat_chair_model = MujocoXMLObject("/home/kai/robosuite/robosuite/models/assets/objects/seat_chair.xml")
world.merge(seat_chair_model)

But, it looks very big in mujoco world. The cotent of the xml file is as shown in the following:
<mujoco model="seat_chair">
  <asset>
    <texture file="../textures/light-wood.png" type="cube" name="light-wood" />
  	<mesh file="meshes/seat_chair.STL" name="seat_chair_mesh"/>
  	<material name="MatLightWood" texture="light-wood" texrepeat="3 3" specular="0.4" shininess="0.1" />
  </asset>
  <worldbody>
    <body>
      <body name="collision">
        <geom pos="0 0 0" mesh="seat_chair_mesh" type="mesh" solimp="0.998 0.998 0.001" solref="0.001 1" density="50" friction="0.95 0.3 0.1" material="MatLightWood" group="1"/>
      </body>
      <site rgba="0 0 0 0" size="0.0001" pos="0 0 -0.045" name="bottom_site"/>
      <site rgba="0 0 0 0" size="0.0001" pos="0 0 0.03" name="top_site"/>
      <site rgba="0 0 0 0" size="0.0001" pos="0.0254 0.0254 0" name="horizontal_radius_site"/>
    </body>
  </worldbody>
</mujoco>

Can anyone give me some advice, Thank you very much!

cremebrule, 2020-12-04:
Hi @HankerSia ,

You can set the scale of your mesh object by setting the `scale` attribute in the corresponding `<mesh ... />` tag. Please see [this reference](http://www.mujoco.org/book/XMLreference.html#mesh) for exact details.

Closing this issue for now, feel free to re-open it if you continue to have problems!

closed 2020-12-04 by cremebrule

HankerSia, 2020-12-04:
> Hi @HankerSia ,
>
> You can set the scale of your mesh object by setting the `scale` attribute in the corresponding `<mesh ... />` tag. Please see [this [...]

Yes, "scale" had been used for adjust the size of the model. Thanks. Another problems, can you give me a demo of grasping the object loaded by xml files, or some reference code demo. Thank you in advance!

cremebrule, 2020-12-04:
I assume this is the same question as the new issue you posted, so I'll answer there.

SuryaPratapSingh37, 2022-01-30:
Hi @HankerSia,                                                                                                                                                                       

Actually, I was loading an object into the robosuite environment using the following code:

[image]

As a result, I do see the object in the environment but it doesn’t seem to be a free body, since it is suspended in space and not experiencing any gravity, even though both the collision and visual mesh are in sync, as can be seen in the attached picture. 
[image]
Secondly, if I provide this MujocoXMLObject position coordinates as if it is sitting on top of a moving BoxObject, which acts as a conveyor, then the MujocoXMLObject doesn’t move with the BoxObject and remains stick to its original position in space. Furthermore, if a second BoxObject is placed on top of the moving conveyor, the second BoxObject does move with the conveyor. In fact, the second BoxObject, if it collides with the MujocoXMLObject, is also not able to push the MujocoXMLObject. I find it pretty weird why the MujocoXMLObject doesn’t show the same behaviour like a BoxObject. So can you please help me fix these issues? I'm also attaching the MujocoXMLObject's collision and visual xml files.
[image]
[image]

Thanks in advance