ARISE-Initiative/robosuite #135: Can anyone give fresh man a simple demo of grasp using rl algorithm(ddpg or pro)?

HankerSia, 2020-12-04:
For I am a fresh man, so I am so confused that how to use robosuite and how to train a grasp agent? 
Can anyone give me a beginner level tutorial. Or tell me how can I create a simple demo of grasp based on robosuite and typical RL algorithm.
Thank you in advance!

amandlek, 2020-12-04:
We've got a [benchmark repository](https://github.com/ARISE-Initiative/robosuite-benchmark) that should allow you to reproduce our RL results that are described in our [whitepaper](https://robosuite.ai/assets/whitepaper.pdf).

closed 2020-12-04 by amandlek

cremebrule, 2020-12-04:
To add to what @amandlek said:

Robosuite in general isn't meant to do any RL on its own -- it's merely an implementation of an "environment" for an RL agent (algorithm) to interact with. We use a gym-style format, so any RL algorithm that works with a Gym environment can be used with our repo (just use the GymWrapper class).

I would recommend starting with [OpenAI's Gym environments / tutorials](https://gym.openai.com/docs/) just to get familiar with the code style -- again, we utilize a similar interface with our environments here.

In terms of RL Algorithms, I would recommend getting familiar with the [stable-baseline repo](https://github.com/hill-a/stable-baselines), which includes many state-of-the-art RL algorithms that you can directly use off-the-shelf with gym-style environments.

I am not sure what your exact background is or your familiarity with RL in general, but I would highly encourage you to become familiar with the underlying theory of RL ([this blog post](https://jonathan-hui.medium.com/rl-introduction-to-deep-reinforcement-learning-35c25e04c199) might be a good and informal starting point) -- once you understand that, understanding RL codebases will become much easier!

HankerSia, 2020-12-04:
> To add to what @amandlek said:
> 
> Robosuite in general isn't meant to do any RL on its own -- it's merely an implementation of an "environment" for an RL agent (algorithm) to interact with. We use a gym-style format, so any RL algorithm that works with a Gym environment can be used with our repo (just use the GymWrapper class). We also have our own [full set of documentation](http://robosuite.ai/docs/overview.html) which might be very helpful.
> 
> I would recommend starting with [OpenAI's Gym environments / tutorials](https://gym.openai.com/docs/) just to get familiar with the code style -- again, we utilize a similar interface with our environments here.
> 
> In terms of RL Algorithms, I would recommend getting familiar with the [stable-baseline repo](https://github.com/hill-a/stable-baselines), which includes many state-of-the-art RL algorithms that you can directly use off-the-shelf with gym-style environments.
> 
> I am not sure what your exact background is or your familiarity with RL in general, but I would highly encourage you to become familiar with the underlying theory of RL ([this blog post](https://jonathan-hui.medium.com/rl-introduction-to-deep-reinforcement-learning-35c25e04c199) might be a good and informal starting point) -- once you understand that, understanding RL codebases will become much easier!

Thank you firstly! I had learned the basic knowledge of RL and gym. A pendulum control experiment using ddpg had been built by myself.  I also read the introduction of the robosuite.  I can build the basic environment which include robot, gripper, and objects and so on.
But, there are not the following reference tutorials about how to use this environment, or a simple code demo like my pendulum control experiment, it only include one python source file...
`
# -*- coding: utf-8 -*-
"""
Created on Sun Nov 29 21:38:39 2020

@author: lenovo
"""

# -*- coding: utf-8 -*-
import os
import random
import gym
from collections import deque

import numpy as np
import tensorflow as tf

from keras.layers import Input, Dense, Lambda, concatenate
from keras.models import Model
from keras.optimizers import Adam
import keras.backend as K

class DDPG():
    """Deep Deterministic Policy Gradient Algorithms.
    """
    def __init__(self):
        super(DDPG, self).__init__()

        self.sess = K.get_session()
        self.env = gym.make('Pendulum-v0')
        self.bound = self.env.action_space.high[0]

        # update rate for target model.
        self.TAU = 0.01
        # experience replay.
        self.memory_buffer = deque(maxlen=4000)
        # discount rate for q value.
        self.gamma = 0.95
        # epsilon of action selection
        self.epsilon = 1.0
        # discount rate for epsilon.
        self.epsilon_decay = 0.995
        # min epsilon of ε-greedy.
        self.epsilon_min = 0.01

        # actor learning rate
        self.a_lr = 0.0001
        # critic learining rate
        self.c_lr = 0.001

        # ddpg model
        self.actor = self._build_actor()
        self.critic = self._build_critic()

        # target model
        self.target_actor = self._build_actor()
        self.target_actor.set_weights(self.actor.get_weights())
        self.target_critic = self._build_critic()
[... 256 lines cut ...]

            x = observation.reshape(-1, 3)
            action = self.actor.predict(x)[0]
            observation, reward, done, _ = self.env.step(action)

            reward_sum += reward

            if done:
                print("Reward for this episode was: {}".format(reward_sum))
                random_episodes += 1
                reward_sum = 0
                observation = self.env.reset()

        self.env.close()

if __name__ == '__main__':
    model = DDPG()

    history = model.train(10, 128)
#    model.save_history(history, 'ddpg.csv')

    model.play()
`

So, if I want train a pick-place task, is there any simple project using robosuite. Current project in github cite all kinds of RL lib...It is hard for me to understand...

amandlek, 2020-12-04:
As mentioned by @cremebrule , we have a simple Gym interface that you can see [here](http://robosuite.ai/docs/demos.html#openai-gym-style-apis). If you have indeed used gym with your own DDPG code before, then you should be able to swap the gym environment you were using with a robosuite environment, by using the wrapper that we've linked. Are there any other points of confusion?

HankerSia, 2020-12-05:
@amandlek Thanks! Can you made a short introduction about the observation and reward of a simple picking task. Then what signals or actions (like torque of the joints of the robotic arm)should be input for this environment.
And how to define an object in the picking task.
`
import os
import random
import gym
from collections import deque
import numpy as np
import tensorflow as tf
from keras.layers import Input, Dense, Lambda, concatenate
from keras.models import Model
from keras.optimizers import Adam
import keras.backend as K
class DDPG():
    """Deep Deterministic Policy Gradient Algorithms.
    """
    def __init__(self):
        super(DDPG, self).__init__()

        self.sess = K.get_session()
        self.env = gym.make('Pendulum-v0')
        self.bound = self.env.action_space.high[0]
        # update rate for target model.
        self.TAU = 0.01
        # experience replay.
        self.memory_buffer = deque(maxlen=4000)
        # discount rate for q value.
        self.gamma = 0.95
        # epsilon of action selection
        self.epsilon = 1.0
        # discount rate for epsilon.
        self.epsilon_decay = 0.995
        # min epsilon of ε-greedy.
        self.epsilon_min = 0.01
        # actor learning rate
        self.a_lr = 0.0001
        # critic learining rate
        self.c_lr = 0.001
        # ddpg model
        self.actor = self._build_actor()
        self.critic = self._build_critic()
        # target model
        self.target_actor = self._build_actor()
        self.target_actor.set_weights(self.actor.get_weights())
        self.target_critic = self._build_critic()
        self.target_critic.set_weights(self.critic.get_weights())
        # gradient function
        self.get_critic_grad = self.critic_gradient()
        self.actor_optimizer()
        if os.path.exists('model/ddpg_actor.h5') and os.path.exists('model/ddpg_critic.h5'):
            self.actor.load_weights('model/ddpg_actor.h5')
            self.critic.load_weights('model/ddpg_critic.h5')
    def _build_actor(self):
        """Actor model.
        """
        inputs = Input(shape=(3,), name='state_input')
        x = Dense(40, activation='relu')(inputs)
        x = Dense(40, activation='relu')(x)
        x = Dense(1, activation='tanh')(x)
        output = Lambda(lambda x: x * self.bound)(x)
        model = Model(inputs=inputs, outputs=output)
        model.compile(loss='mse', optimizer=Adam(lr=self.a_lr))
        return model
    def _build_critic(self):
        """Critic model.
        """
        sinput = Input(shape=(3,), name='state_input')
        ainput = Input(shape=(1,), name='action_input')
        s = Dense(40, activation='relu')(sinput)
        a = Dense(40, activation='relu')(ainput)
[... 166 lines cut ...]
        return history
    def play(self):
        """play game with model.
        """
        print('play...')
        observation = self.env.reset()
        reward_sum = 0
        random_episodes = 0
        while random_episodes < 10:
            self.env.render()
            x = observation.reshape(-1, 3)
            action = self.actor.predict(x)[0]
            observation, reward, done, _ = self.env.step(action)
            reward_sum += reward
            if done:
                print("Reward for this episode was: {}".format(reward_sum))
                random_episodes += 1
                reward_sum = 0
                observation = self.env.reset()
        self.env.close()
if __name__ == '__main__':
    model = DDPG()
    history = model.train(10, 128)
    model.play()
`