ARISE-Initiative/robosuite #159: object document update

wangcongrobot, 2021-01-07:
Hi, it seems that some documents do not update.

In the [quickstart doc](https://github.com/wangcongrobot/robosuite/blob/master/docs/quickstart.md), cannot find function `get_collision`
```python
sphere = BallObject(
    name="sphere",
    size=[0.04],
    rgba=[0, 0.5, 0.5, 1]).get_collision()
```
And in [object doc](https://github.com/wangcongrobot/robosuite/blob/master/docs/modules/objects.md), the two functions cannot be found:
```python
    def get_collision(self, name=None, site=False):
        pass

    def get_visual(self, name=None, site=False):
        pass
```

cremebrule, 2021-01-07:
Thanks for bringing this up. Yes, the docs are in the process of being updated, and should be finalized and published by this weekend.

cremebrule, 2021-02-18:
This is now fixed!

closed 2021-02-18 by cremebrule