ARISE-Initiative/robosuite #160: Difficulty installing on MacOS 10.15.7 with python 3.7.3 && pip 20.3.3

quinnliu, 2021-01-10:
I'm not able to install robosuite as I keep getting stuck on the install step `pip install robosuite` here: https://robosuite.ai/docs/installation.html

Any advice would be appreciated:

These are the steps I've done so far: 

1. MacOS 10.15.7

2. `chmod 755 getid_osx`

3. System Preferences -> Security & Privacy -> General ->
lower section "Allow apps downloaded from:" pop ups

4. `mv mujoco200_macos/ mujoco200` otherwise you will get error:

5. current folder && file structure = 
~/
- .mujoco/
  - mjkey.txt 
  - mujoco200/
    - bin/
    - doc/
    - include/
    - model/
    - sample/

6. running `pip install robosuite` says the mac is missing gcc && suggests running `brew install gcc`

7. This brew command isn't working so I'm following all of the steps here: 
https://github.com/openai/mujoco-py#youre-on-macos-and-you-see-clang-error-unsupported-option--fopenmp

which does remove the error message about gcc

8. running  pip install robosuite now gives the error `ERROR: Could not build wheels for mujoco-py which use PEP 517 and cannot be installed directly`

quinnliu, 2021-01-10:
11. `brew install --build-from-source llvm` 
12. `xcode-select --install`

quinnliu, 2021-01-10:
13. I'm noticing that `pip install robosuite` uses mujoco_py 2.0.2.9 while I do have mujoco_py 2.0.2.8 installed successfully

quinnliu, 2021-01-10:
14. After checking https://github.com/openai/mujoco-py/releases I'm also confused why robosuite is using version 2.0.2.9 when 2.0.2.5 is the latest release? Thanks!

cremebrule, 2021-01-10:
Hi @quinnliu ,

Yes, unfortunately mujoco can be a bit fickle when trying to install. I too had this problem awhile back.

Mujoco 2.0.2.8 should be fine; feel free to change the requirement in `requirements.txt`. We include 2.0.2.9 since that is the version that I have been testing robosuite on.

I believe in order to get mujoco installed, I had to force a specific (older) gcc to be used when building mujoco. However, apparently [the commit from last year](https://github.com/openai/mujoco-py/commit/d73ce6e91d096b74da2a2fcb0a4164e10db5f641) had fixed this issue, which is another reason why mujoco 2.0.2.X was required. But given that mujoco must be built before installing mujoco-py, I suppose that might not help too much.

Have you checked doing some of the proposed fixes in [this thread](https://github.com/openai/mujoco-py/issues/463)? I believe this is the reference I had used to get mujoco working originally when I ran into your similar issue.

quinnliu, 2021-01-10:
Hi @cremebrule 

Thanks for the quick response. I tried the gcc suggestions && it's still not working. Trying the renaming folders strategy however this does not seem like a safe or scalable solution for all future users...

I'm still confused as to how you are able to use mujoco 2.0.2.9 when their public GitHub only released version 2.0.2.5 Is there a private git repo with more up to date releases I'm not aware of? Thanks!

~ Q

quinnliu, 2021-01-10:
@cremebrule I've tried all of the strategies on MacOS 10.15.7 inside a python 3.7.3 virtual environment with pip 20.3.3 && nothing has worked. 

Any other ideas? I would prefer to get this working on MacOS Catalina 10.15.7 than linux

cremebrule, 2021-01-10:
Hi @quinnliu ,

Hmmmm, that is concerning. I do know a lot has changed since pip moved to version 20; perhaps downgrading pip to version ~18 or so might perform differently?

Regarding mujoco version, I'm not sure how the devs are handling their public release notes, but pip draws from pypi, which currently has mujoco-py already updated to 2.0.2.13 (see [HERE](https://pypi.org/project/mujoco-py/)).

quinnliu, 2021-01-10:
@cremebrule I have it installed successfully with the following steps

# How to install Robosuite in less than 5 hours on MacOS 10.15.7
1. MacOS 10.15.7 && Python 3.7.3 virtual environment with Pip
20.3.3

2. https://robosuite.ai/docs/installation.html

3. `chmod 755 getid_osx`

4. `mv mujoco200_macos/ mujoco200` otherwise you will get error on `pip install robosuite`

5. current folder && file structure =
~/
- .mujoco/
  - mjkey.txt <---- 1st HERE
  - mujoco200/
    - bin/
      - mjkey.txt <---- ALSO HERE
    - doc/
    - include/
    - model/
    - sample/

6. System Preferences -> Security & Privacy -> General ->
lower section "Allow apps downloaded from:" pop ups

7. `brew uninstall gcc` all versions like
  - `brew uninstall gcc@6`
  - `brew uninstall gcc@7`
  - `brew uninstall gcc@8`
  - `brew uninstall gcc@9`

8. `brew install llvm boost hdf5`

9. In your `.bashrc`

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/qliu/.mujoco/mujoco200/bin # <------ replace qliu with ur output of command
                                                                          # `pwd` in
                                                                          # folder mujoco200/bin
export PATH="/usr/local/opt/llvm/bin:$PATH"

export CC="/usr/local/opt/llvm/bin/clang"
export CXX="/usr/local/opt/llvm/bin/clang++"
export CXX11="/usr/local/opt/llvm/bin/clang++"
export CXX14="/usr/local/opt/llvm/bin/clang++"
export CXX17="/usr/local/opt/llvm/bin/clang++"
export CXX1X="/usr/local/opt/llvm/bin/clang++"

export LDFLAGS="-L/usr/local/opt/llvm/lib"
export CPPFLAGS="-I/usr/local/opt/llvm/include"
```

10. source `.bashrc`

11. in python 3.7.3 virtual environment
`pip install -U 'mujoco-py<2.1,>=2.0'`

12. `python -c "import mujoco_py"`

13. `pip freeze > requirements.txt` && open requirements.txt should
look exactly like:

```
bcrypt==3.2.0
cffi==1.14.4
Cython==0.29.21
fasteners==0.16
glfw==2.0.0
imageio==2.9.0
mujoco-py==2.0.2.13
numpy==1.19.5
Pillow==8.1.0
pycparser==2.20
six==1.15.0
```

14. `pip3 install robosuite`

cremebrule, 2021-01-10:
Hi @quinnliu ,

Thanks for taking the time to debug, this is great information! We will add this to the new docs that will be published shortly in a couple of days.

Closing this issue.

closed 2021-01-10 by cremebrule