ARISE-Initiative/robosuite #161: what am I doing wrong in the build your own environment example

quinnliu, 2021-01-11:
Hi, 

Really excited to lay a virtual wall of bricks using robosuite API. 

In the build your own environment example I'm getting an error on the sphere BallObject

```
from robosuite.models import MujocoWorldBase
from robosuite.models.robots import Panda
from robosuite.models.grippers import gripper_factory
from robosuite.models.arenas import TableArena
from robosuite.models.objects import BallObject
from robosuite.utils.mjcf_utils import new_joint
from mujoco_py import MjSim, MjViewer

world = MujocoWorldBase()

mujoco_robot = Panda()

gripper = gripper_factory('PandaGripper')
mujoco_robot.add_gripper(gripper)

mujoco_robot.set_base_xpos([0, 0, 0])
world.merge(mujoco_robot)

mujoco_arena = TableArena()
mujoco_arena.set_origin([0.8, 0, 0])
world.merge(mujoco_arena)

sphere = BallObject(
    name="sphere",
    size=[0.04],
    rgba=[0, 0.5, 0.5, 1]).get_collision()
sphere.append(new_joint(name='sphere_free_joint', type='free'))
sphere.set('pos', '1.0 0 1.0')
world.worldbody.append(sphere)
```

error
```
  File "build_wall.py", line 52, in <module>
    rgba=[0, 0.5, 0.5, 1]).get_collision()
AttributeError: 'BallObject' object has no attribute 'get_collision'
```

I'm not sure what I'm doing wrong as I'm just copying the example code. Was the 
get_collision() function deprecated? If so how would I figure out exactly which functions I can && can not call? 

Thanks,
Q

cremebrule, 2021-01-11:
Hi @quinnliu ,

Yes, unfortunately, the docs have not been updated since v1.0.0. We are currently on v 1.1, which heavily refactored the backend to make prototyping much easier, and v1.2 will be released in a few days, along with an updated version of the docs. Apologies for the delay!

In the meantime, to get your code working, you can instead call `.get_obj()` (instead of `get_collision`, which indeed has been depreceated) to get the xml to append to your mujoco world object.

cremebrule, 2021-01-11:
I should also note that all the source code has been heavily documented, if you are interested in digging through the exact functionality currently present. The [changelog for v1.1](https://github.com/ARISE-Initiative/robosuite/releases/tag/v1.1) also covers some of these changes. Otherwise, stay tuned for the v1.2 release with updated docs in a few days! :)

quinnliu, 2021-01-11:
@cremebrule Thank you very much! The change log for v1.1 should be good for now :)

cremebrule, 2021-01-11:
The `robosuite/demos` folder accessed from this repo should have all the updated demo scripts.

closed 2021-01-12 by yukezhu