ARISE-Initiative/robosuite #164: why is the world.get_model(mode="mujoco_py") call not working

quinnliu, 2021-01-12:
I looked through the code in robosuite/models/base.py", line 130

I'm confused as to why the sphere example code is stuck here: 

Any advice would be appreciated

```
sphere = BallObject(
    name="sphere",
    size=[0.04],
    rgba=[0, 0.5, 0.5, 1]).get_obj()
sphere.append(new_joint(name='sphere_free_joint', type='free'))
sphere.set('pos', '1.0 0 1.0')
world.worldbody.append(sphere)

model = world.get_model(mode="mujoco_py")
```

error message: 
```
  File "build_wall.py", line 71, in <module>
    model = world.get_model(mode="mujoco_py")
  File "/Users/qliu/Documents/agihome.com/robosuite_env/lib/python3.7/site-packages/robosuite/models/base.py", line 130, in get_model
    model = load_model_from_xml(string.getvalue())
  File "mujoco_py/cymj.pyx", line 191, in mujoco_py.cymj.load_model_from_xml
Exception: <mujoco model="base">
  <compiler angle="radian" inertiagrouprange="0 0" meshdir="meshes/" />
  <option cone="elliptic" impratio="20" />
  <size nconmax="5000" njmax="5000" />
```

quinnliu, 2021-01-12:
update: I realized I was reading the `master` branch && not `v1.1` my bad

cremebrule, 2021-01-12:
I'm not entirely sure -- that error message is cryptic. One bug I noticed is that you try to attach another free joint to the sphere -- however, there's already a free joint that is enabled by default. Can you remove that line and try running the code again?

quinnliu, 2021-01-12:
@cremebrule Thanks your right about the sphere.append()

closed 2021-01-12 by quinnliu