ARISE-Initiative/robosuite #165: how to create 12 "brick"s for the Panda arm to build into a wall?

quinnliu, 2021-01-12:
I've read through all the documentation on branch v1.1 for 
env = suite.make(...) function && it's not clear to me how I would add 12 "bricks" 
to the environment

In the example of 

env = suite.make(
    env_name="Lift", # "Stack", "Door"
    robots="Panda",  # "Sawyer" and "Jaco"
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
)

a singular cube is seemingly auto generated && I can't find where it is explicitly declared

If I wanted to add 12 more bricks for the arm to stack into a wall how would I do this exactly?

Would it look something like the following?

quinnliu, 2021-01-12:
I found the hammer.py file in robosuite.

Can I code a new file brick.py && square_window.py in my own project folder outside of robosuite/ && add it to my robot environment so that when I call `env.render()` I'll see Pandas arm, 12 bricks, && 1 square window?

cremebrule, 2021-01-12:
This shouldn't be a problem. An even quicker way would be to leverage our built-in `BoxObject` as individual "bricks".

Closing this issue, feel free to re-open if you run into any further issues.

closed 2021-01-12 by cremebrule