ARISE-Initiative/robosuite #168: ModuleNotFoundError: No module named 'robosuite.utils.observables'

hermanjakobsen, 2021-01-12:
Hi guys! 

I tried to update my robosuite package to v1.1.0 using `pip install robosuite -U`. To test out the new version I tried to simulate the lift environment. However, I am getting an import error that the `observables.py` file is missing. Checking out the robosuite package installed in my virtual environment, the file is indeed missing.

Have I done something wrong when upgrading the package?

hermanjakobsen, 2021-01-12:
I have also found additional inconsistencies in how the environment classes are implemented in the installed package versus the implementations here on Github.

yukezhu, 2021-01-12:
Hi @hermanjakobsen Thanks for reaching out to us regarding this inconsistency. Indeed, the current master HEAD is ahead of the pip package (which is currently the v1.10 release). The master branch contains some new changes of the sensor APIs (`obeservables`) that are useful for domain/dynamics randomization. We have two pending updates to complete before we update both pip and the master HEAD to v1.20. If you want to check out the checkpoint of the current pip version, here you can find the link: https://github.com/ARISE-Initiative/robosuite/releases/tag/v1.1

closed 2021-01-25 by yukezhu