ARISE-Initiative/robosuite #169: possible to create an camera for input into a visual cognitive computer in the robot simulation environment?

quinnliu, 2021-01-12:
To generate the array of 8 numbers stored as the variable `actions` for the Panda 8 DOF robotic arm to move towards a high level goal I'm curious how can one generate a visual 
input camera inside the environment simulation? 

In the real world this is required for the Panda arm to move correctly but I'm confused as to how to do this in the simulation. Thanks!

cremebrule, 2021-01-12:
Sorry, I'm not sure exactly what is being asked -- are you asking if you can receive image observations from the environment for a policy to generate actions from? If so, we already have that functionality enabled -- please see [this guide](http://robosuite.ai/docs/modules/environments.html#interacting-with-an-environment) for more info (slightly outdated in terms of naming specifics, but the high-level infra is consistent)

quinnliu, 2021-01-12:
Sorry for the confusion.

To be more specifically I'm wondering if I can process the RGB rendered image of the 3D environment as part of my 
input to process to generate the newest `action` array variable 

For example:

```
env = suite.make(
    env_name="TwoArmLift", # Lift, Stack, NutAssembly, NutAssemblySingle, NutAssemblySquare, NutAssemblyRound, PickPlace, PickPlaceSingle, PickPlaceMilk, PickPlaceBread, PickPlaceCereal, PickPlaceCan, Door, Wipe, TwoArmLift, TwoArmPegInHole, TwoArmHandover
    robots=["Panda", "Panda"],  # Sawyer, Jaco, Panda
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=True,
)

# how to place a RGB camera in 3D simulation to use as input to determine `action`
# where x,y,z are relative to robot arm base
env.add_stationary_camera(x=0, y=8, z=0, resolution="1080x1080", type="rgb")

# or attach a camera relative to the end effector gripper
env.add_dynamic_camera(x=0, y=1, z=0, resolution="1080x1080", type="rgb")

env.reset()

for i in range(1000):
   # get current time step camera output 
   2D_array_representing_3D_scene = env.get_view_at_timestep(i)

   # process 2D_array_representing_3D_scene using a visual cognitive computer 
   # to generate 2D array of actions to pass into env.step(action) instead of 
   # passing a random array

    action = np.random.randn(env.robots[0].dof*2) # sample random action
    print("action = ", action)
    # action =  [ 0.39813241  0.90703087  0.97952504 -0.07412093  1.05065751  0.96402113
    # -1.4640924   0.57175383]
    ordered_dict, reward, is_done, info_dict = env.step(action)  # take action in the environment
    env.render()  # render on display
```

quinnliu, 2021-01-12:
[image]

quinnliu, 2021-01-12:
# POV 2
[image]

quinnliu, 2021-01-12:
# POV 3
[image]

quinnliu, 2021-01-12:
NVM: I figured it out as you can pass arguments to sim.render(...)

closed 2021-01-12 by quinnliu