ARISE-Initiative/robosuite #172: Anyone interested in coding the robot arms to build a wall of bricks for a home?

quinnliu, 2021-01-14:
Hi Dr. Zhu && Josiah,

Not sure if this is okay to post here but I'm very very determined to build a real home using generally intelligent robotics. If this isn't okay feel free to close the issue :) 

Specifically I'm interested in coding a simulation using `robosuite` of a visual cognitive computer(VCC) from papers here:
1. https://vcrs.wpengine.com/wp-content/uploads/2020/03/10.1126@scirobotics.aav3150.pdf
2. https://vcrs.wpengine.com/wp-content/uploads/2020/03/behavior-is-everything-pdf.pdf
3. https://science.sciencemag.org/content/sci/358/6368/eaag2612.full.pdf

And then using the VCC to build a home out of bricks using their IKEA image training input strategy.

After we figure out how to lay the bricks in the `robosuite` simulation I plan to take it a lot further like windows, doors, plumbing, sink, toilet, shower... && then using the real Panda arm for building a tiny home at 1/12 scale && then at real home scale. 

But I think it's important to start small with a wall of bricks in simulation.

Best,
Q

quinnliu, 2021-01-14:
I'm just getting started && this is my progress so far :) 

[image]

yukezhu, 2021-01-14:
Hi @quinnliu This issues page is dedicated to bug reports and feature requests. You are welcome to discuss your ideas of using robosuite on other online platforms, but this is not the right place. Thanks.

closed 2021-01-14 by yukezhu