ARISE-Initiative/robosuite #173: Playback script does not reproduce deterministic results

zhuyifengzju, 2021-01-14:
Currently the scripts for playing back demonstrations seem to have a problem to reproduce the same trajectories as in the collected demonstrations. So if first collect data using:
`python scripts/collect_human_demonstrations.py --device spacemouse`, and then run  `python scripts/playback_demonstrations_from_hdf5.py --folder DEMONSTRATION_FOLDER --use-actions`, it will gives the following error:
`Traceback (most recent call last):
  File "scripts/playback_demonstrations_from_hdf5.py", line 93, in <module>
    assert(np.all(np.equal(states[j + 1], state_playback)))
AssertionError`

Seems like the deterministic playback is broken somehow.

yukezhu, 2021-01-14:
Confirmed this issue on my end. Example [demo.hdf5](https://utexas.box.com/s/tjlnuec4e3xzjcoalkk9mq8odn3j3l5c)

State-only playback
[image]

Action playback
[image]

yukezhu, 2021-01-14:
Also, playback script complained about missing `joint_torques` values in hdf5
https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/scripts/playback_demonstrations_from_hdf5.py#L83

closed 2021-01-15 by yukezhu