ARISE-Initiative/robosuite #176: Change robot's sensor frequency to higher frequency?

zhuyifengzju, 2021-01-15:
The current sensor frequency for robot's states are by default the same as the robot's control frequency : https://github.com/ARISE-Initiative/robosuite/blob/a18f1a79a4baa612c43917efd13ff299ff19e7ca/robosuite/robots/robot.py#L220

I propose to change the sampling rate variable to another variable so that we have control over changing the sampling frequency for proprioceptive states. I think it is desirable that we get proprioceptive states faster than images.

yukezhu, 2021-01-27:
We had a discussion on this with @cremebrule . The current `observable` APIs support buffering sensory readings of a higher frequency than the control frequency. The way to achieve this is through the use of [filter](https://github.com/ARISE-Initiative/robosuite/blob/779a3b7c91e8faab4a4f1285acae3e8c9b63faaf/robosuite/utils/observables.py#L141). However, the additional documentations/demo examples could be beneficial to illustrate such use cases. This will be added in our next version release.

closed 2021-01-27 by yukezhu

cremebrule, 2021-01-30:
To follow up with this, I've provided an example script below that concretely showcases how the `filter` functionality of the `Observable` class can easily be used to record values as they are sampled in real (sim) time. This will also be added to our documentation in the upcoming release.

```python
import robosuite as suite
import numpy as np
from robosuite.utils.buffers import RingBuffer

# Create env instance
control_freq = 10
env = suite.make("Lift", robots="Panda", has_offscreen_renderer=False, use_camera_obs=False, control_freq=control_freq)

# Define a ringbuffer to store joint position values
buffer = RingBuffer(dim=env.robots[0].robot_model.dof, length=10)

# Create a function that we'll use as the "filter" for the joint position Observable
# This is a pass-through operation, but we record the value every time it gets called
# As per the Observables API, this should take in an arbitrary numeric and return the same type / shape
def filter_fcn(corrupted_value):
    # Record the inputted value
    buffer.push(corrupted_value)
    # Return this value
    return corrupted_value

# Now, let's enable the joint position Obervable with this filter function
env.modify_observable(
    observable_name="robot0_joint_pos",
    attribute="filter",
    modifier=filter_fcn,
)

# Let's also increase the sampling rate to showcase the Observable's ability to update multiple times per env step
obs_sampling_freq = control_freq * 4
env.modify_observable(
    observable_name="robot0_joint_pos",
    attribute="sampling_rate",
    modifier=obs_sampling_freq,
)

# Take a single environment step with positive joint velocity actions
arm_action = np.ones(env.robots[0].robot_model.dof) * 1.0
gripper_action = [1]
action = np.concatenate([arm_action, gripper_action])
env.step(action)

# Now we can analyze what values were recorded
np.set_printoptions(precision=2)
print(f"\nPolicy Frequency: {control_freq}, Observable Sampling Frequency: {obs_sampling_freq}")
print(f"Number of recorded samples after 1 policy step: {buffer._size}\n")
for i in range(buffer._size):
    print(f"Recorded value {i}: {buffer.buf[i]}")
```

JoJo-Create, 2023-05-23:
Hi, I try to maintain a buffer to store the ft data between two env.step(), here is what I did. 
```python
@sensor(modality=modality)
def ee_forces_continuous(obs_cache):
        if "ee_forces_continuous" not in obs_cache:
            obs_cache["ee_forces_continuous"] = RingBuffer(dim = 3, length=self.ft_buffer_size)
        else:
            obs_cache["ee_forces_continuous"].push(self.robots[0].ee_force - self.ee_force_bias)
        return obs_cache["ee_forces_continuous"].buf

sensors += [ee_forces_continuous]
names += [f"{pf}ee_forces_continuous"]

for name, s in zip(names, sensors):
        observables[name] = Observable(
            name=name,
            sensor=s,
            sampling_rate=self.control_freq,
            )
        if name == f"{pf}ee_forces_continuous":
            observables[name].set_sampling_rate(int(self.control_freq * self.ft_buffer_size))
    return observables
```
When I set the ''self.ft_buffer_size'' higher then 10, then I get the warning: "sampling rate for observable robot0_ee_forces_continuous is either too low or delay is too high. Please adjust one (or both)". But the data seems just fine. Can I just ignore this warning? I also wonder how often the "self.robots[0].ee_force" gets updated

JoJo-Create, 2023-05-23:
> Hi, I try to maintain a buffer to store the ft data between two env.step(), here is what I did.
>
> ```python
> @sensor(modality=modality)
> def ee_forces_continuous(obs_cache):
>         if [...]

I think the sampling rate was higher than the internal update rate. When I decrease SIMULATION_TIMESTEP to 0.001 in macros_private.py or decrease control_freq to 5, the warning disappears