ARISE-Initiative/robosuite #182: Converting Observations from Human Demos to OpenAI observations

IzzyPutterman, 2021-01-28:
I recently collected some human demos (using the script) on the Door env using the UR5e robot and all other default settings and got observations of the shape (x, 29) and actions of the shape (x, 7).  I am trying to train my model which is built around the Open AI gym setup.  The observation space in your OpenAI gym wrapper has a shape of (51,) and an action space shape of (7,).  The actions clearly line up, but I was wondering if there was an easy way to cast my human demo observations into gym observations.

I have been looking through the code base and I don't see an obvious way to do this (may have missed something).  Any workflow for using human demos with open AI gym would be much appreciated.

cremebrule, 2021-01-28:
Hi @IzzyPutterman ,

What observation space were you using when collecting demos? That seems quite small for the observation space. It's difficult for me to infer what settings you used, but my initial thought is that you had set `use_camera_obs=False` and `use_object_obs=False`. In fact, 29 corresponds to the robot-specific observations for the UR5e.

IzzyPutterman, 2021-01-28:
Hey,

Yes the use_camera_obs=False.  Not sure about use_object_obs. I just used the default settings (except for the robot and env) on the Collect_human_demons script.  I'll try to recollect the demos with this change.

IzzyPutterman, 2021-01-28:
I looked into it a bit more, seems like both the env im using to collect the demos and the Open AI env I'm creating both have use_camera_obs=False.  I also did a check and found that the instance of the data_collection_wrapper I am initializing (the one with obs of size (29,), has use_object_obs=True. 

Both the OpenAI gym env and human demo env have use_camer_obs=False and use_object_obs=True.
Is there some obvious observation I am missing?

amandlek, 2021-01-28:
I think the main issue is that your demonstrations have the mujoco simulation state, but you would like the environment observations. You can use a similar script to [our playback script](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/scripts/playback_demonstrations_from_hdf5.py#L104) for post-processing the demonstration data, and extracting the actual observations from the environment - you can call `env._get_observations(force_update=True)` after setting the state to retrieve those. Note that there is currently a bug that will be fixed when we merge [this commit](https://github.com/ARISE-Initiative/robosuite/pull/151/commits/007ddeac78285337110755a31f767f16c79b430d) into master (which will happen in a few days).

closed 2021-01-28 by amandlek