ARISE-Initiative/robosuite #184: Load objects from URDF files

hermanjakobsen, 2021-02-04:
Hi, guys! 

While searching around the internet I have found some objects that I would like to include in my environment. However, these objects are specified in the URDF file format.

Is it possible to load objects from URDF files in the robosuite framework? I have looked through `objects.py` and `models/base.py`, and from what I've seen I guess the answer is no. It looks like models can only be loaded from MJCF files, If so, is there an easy way to convert from URDF to MJCF?

cremebrule, 2021-02-04:
Hi @hermanjakobsen ,

Unfortunately, we do not support direct loading of URDF files...this may be a feature we'll consider for a future release!

In the meantime, we have a [rudimentary guide](https://docs.google.com/document/d/1bSUKkpjmbKqWyV5Oc7_4VL4FGKAQZx8aWm_nvlmTVmE/edit?userstoinvite=zilmano@gmail.com&ts=5fc7d5bb&actionButton=1) on how to import raw Robot / Gripper models (based on a URDF source file) into robosuite -- I think this can easily be applied to your use case. I would recommend following the initial relevant steps to get an imported MJCF XML file and then modify the XML directly so that its structure matches that of other XML objects already in robosuite. Feel free to post any issues you have along the way.

cremebrule, 2021-02-18:
Closing for now, feel free to re-open or start a new issue

closed 2021-02-18 by cremebrule