ARISE-Initiative/robosuite #187: About weird behaviors for cylinder object in stack env

YunchuZhang, 2021-02-10:
Hi there,
I just try to change the cube object to cylinder like this in staking environment :
[image]
But after that, the cylinder keep weird actions itself without any contact with gripper like rotate itself, and jump down to the floor:
[image]
[image]
[image]
[image]

Do you know why this happened? Thanks

zhuyifengzju, 2021-02-10:
Have you defined the  sampler properly for the cylinder (both position and orientation, z_offset etc.)?

YunchuZhang, 2021-02-10:
[image]
do you mean this? I did not change the original code. Just replace the cube B with a cylinder

zhuyifengzju, 2021-02-11:
try to increase `z_offset`. I assume currently the cylinder is initialized to penetrate the table, and creating large force on the object.

YunchuZhang, 2021-02-11:
I change the z_offset to 0.1, but it still has the weird performance

YunchuZhang, 2021-02-11:
I also troed CapsuleObject, BallObject
and those works well except the cylinder

cremebrule, 2021-02-11:
Hi, I have been able to reproduce your issue. Will look into this to see if there's an easy fix.

YunchuZhang, 2021-02-12:
> Hi, I have been able to reproduce your issue. Will look into this to see if there's an easy fix.

I tried to parse  solimp = [0.998, 0.998, 0.001], solref = [0.001, 1],
but still does not work

yukezhu, 2021-04-27:
This issue has been resolved in PR #213

closed 2021-04-27 by yukezhu