ARISE-Initiative/robosuite #189: roboturk sim dataset can't be performed successfully within current version robosuite

xtkisscs, 2021-02-14:
Hi, I am currently doing some experiments by using robosuite and roboturk sim datasets, but the roboturk sim dataset can only be performed successfully within the version from last year(right, I try to fix this problem and git checkout to the old version robosuite then the problem is fixed), so could I please to ask if the roboturk sim datasets are still compatible with the current version robosuite and how do I use it under the current version robosuite?

weidi-chang, 2021-02-17:
Same question here! 

Are there plans to release the roboturk sim dataset reformatted for v1.0/1.1 in the near future?/Any way to help with the reformatting process?

amandlek, 2021-02-18:
Great question! Unfortunately our roboturk pilot dataset is only compatible with v0.3 and does not directly work with v1.0+. There are indeed ways that these demonstrations could be "reformatted" but they would be approximate due to the large simulation dynamics changes from v0.3 to v1.0+ - for example one scheme might be to try and use trajectory following mechanisms to have the robot follow the same path from the v0.3 demonstrations. 

This is rather involved, so we are opting to collect new datasets that will be compatible with v1.2+. We will release new datasets in the near future, but for now we have released a small sample of data - check [this link](http://robosuite.ai/docs/algorithms/roboturk.html) for more information.

closed 2021-02-18 by amandlek