ARISE-Initiative/robosuite #193: Running into error when making environment

awesome-aj0123, 2021-02-24:
I am currently trying to make the TwoArmLift environment, however, I am running to the following error when doing so:

**Traceback (most recent call last):**
  File "C:\Abhi\Python Projects\robosuite-dev\robosuite\renderers\visii\visii_render_wrapper.py", line 369, in <module>
    control_freq=10, 
  File "c:\abhi\python projects\robosuite\robosuite\environments\base.py", line 41, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "c:\abhi\python projects\robosuite\robosuite\environments\manipulation\two_arm_lift.py", line 200, in __init__
    camera_depths=camera_depths,
  File "c:\abhi\python projects\robosuite\robosuite\environments\manipulation\manipulation_env.py", line 168, in __init__
    robot_configs=robot_configs,
  File "c:\abhi\python projects\robosuite\robosuite\environments\robot_env.py", line 196, in __init__
    hard_reset=hard_reset,
  File "c:\abhi\python projects\robosuite\robosuite\environments\base.py", line 138, in __init__
    self._load_model()
  File "c:\abhi\python projects\robosuite\robosuite\environments\manipulation\two_arm_lift.py", line 299, in _load_model
    robot.robot_model.set_base_ori(rot)
  File "c:\abhi\python projects\robosuite\robosuite\models\robots\robot_model.py", line 87, in set_base_ori
    rot = mat2quat(euler2mat(rot))[[3,0,1,2]]
TypeError: expected dtype object, got 'numpy.dtype[float32]'

It seems that the method `euler2mat` is correctly returning a 3x3 matrix, but when it goes to pass it into the `mat2quat` function, it gives this error. I also tried to run with different environments, but I land into similar issues. Does anybody know why this might be happening and how I can go about fixing it?

amandlek, 2021-02-24:
thanks for bringing this up - can you please make a minimal test script to reproduce the issue on the latest master branch commit?

awesome-aj0123, 2021-02-24:
```
import numpy as np
import robosuite as suite

# create environment instance
env = suite.make(
	"TwoArmLift",
	robots = ["Sawyer", "Panda"],
	reward_shaping=True,
	has_renderer=True,            # no on-screen renderer
	has_offscreen_renderer=False, # no off-screen renderer
	ignore_done=True,
	use_object_obs=True,          # use object-centric feature
	use_camera_obs=False,         # no camera observations
	control_freq=10,
)

# reset the environment
env.reset()

for i in range(200):
    action = np.random.randn(16)  # sample random action
    obs, reward, done, info = env.step(action)  # take action in the environment
    env.render()  # render on display
```

awesome-aj0123, 2021-02-24:
I think the issue stems from a compatibility issue with numpy. Originally, I had v1.20 for numpy, but after reverting it back to v1.19.5, the program works fine. I am not sure why this is happening though.

cremebrule, 2021-02-24:
What version numba are you using? [This thread](https://github.com/numba/numba/issues/6041) suggests that numba may have something to do with this.

A quick sanity check would be to set `ENABLE_NUMBA=False` in `robosuite/utils/macros.py` and see if the problem goes away.

awesome-aj0123, 2021-02-24:
Yes, now with numpy v1.20.1 and after setting `ENABLE_NUMBA=False`, the program works.

cremebrule, 2021-02-24:
Can you upgrade to the newest numba and try seeing if the program still crashes when `ENABLE_NUMBA=True`?

awesome-aj0123, 2021-02-24:
Yes, the program works after upgrading numba with `ENABLE_NUMBA=True`.

closed 2021-02-25 by awesome-aj0123