ARISE-Initiative/robosuite #195: How to add more cameras in environment or modified current camears?

YunchuZhang, 2021-02-26:
As title,
I found there is camera_names called frontview and agentview
where are those defiened? And How can I add more cameras?
Also, If I want to add multi-camera rgb-d in observation space, is that possible?

Thanks

YunchuZhang, 2021-02-26:
Btw, could we use the onscreen and offscreen renderers together like rendering while collecting rgbd data

cremebrule, 2021-02-27:
Hi @YunchuZhang ,

Cameras are natively defined in the mujoco XML file that procedurally generated before the simulation begins. You can access the camera names via `env.sim.model.camera_names`, and can add a new camera using robosuite's [Arena object API](http://robosuite.ai/docs/modeling/arena.html#robosuite.models.arenas.arena.Arena.set_camera) as well (note that any new cameras you add must be added before the environment model is actually loaded into sim!)

You unfortunately cannot use the onscreen and offscreen renderers simultaneously (we have a warning in place as well to prevent attempts at this); a common workaround is the copy the RGB frame(s) that you receive from the obs dict and use those for onscreen rendering (e.g., using pygame or some other renderer).

Closing the issue for now, feel free to re-open if you have further problems.

closed 2021-02-27 by cremebrule