ARISE-Initiative/robosuite #197: How to close the visualization of End effector visualization

YunchuZhang, 2021-03-01:
As title: could you ping me where is the flag to close this green line during offline renderer and online renderer
[image]

Thanks

cremebrule, 2021-03-01:
I assume you are using a `VisualizationWrapper`, else such visualization would not be enabled to begin with.

You can toggle gripper visualization by using the wrapper's built-in setter function:

```python
env.set_visualization_setting(setting="grippers", visible=False)
```

For reference, see [this section](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/environments/manipulation/manipulation_env.py#L172) of the base manipulation env class, which highlights the added visualization keys added for that env. In general, you can use `env.get_visualization_settings()` to get all the different visualization settings that can be toggled using the above setter function.

closed 2021-03-01 by cremebrule

YunchuZhang, 2021-03-01:
Actually I did not use VisualizationWrapper so I did not find the get_visualization_settings functions.
I use offline renderer to render the rgb observation I get from cameras. But it suddenly has the gripper green visualizations.
Is there a way to close it?

YunchuZhang, 2021-03-01:
@cremebrule just to make sure you can see the closed issue reply